Tympan/Tympan_Library #18: Some examples #define POT_PIN, why?

biomurph, 2019-07-08:
Had to correct multiple examples that were defining the wrong POT_PIN number.
Is there a reason that some sketches have this #define? The other examples all use the library readPotentiomenter function call.

chipaudette, 2019-07-08:
The only reason POT_PIN still exists in some sketches is that they've not yet been updated to the new approach (ie, using the Tympan class and its associated "readPotentiomenter()" method).

All sketches should be updated to use the Tympan class instead.  It's my fault that some were missed.

biomurph, 2019-07-08:
Alright, cool. When I have time I will look in and see if there are still outstanding I can knock them out pretty quick.

closed 2019-07-08 by biomurph