Tympan/Tympan_Library #20: failed build with 05-FullSystems/* example sketches, one error and others too large

shapr, 2019-08-04:
I'm using Ubuntu 19.04 with Arduino 1.8.9 and Teensyduino 1.46
and the very latest git clone of Tympan_Library

I've failed to compile all of the 05-FullSystems/*

* WDRC_8BandFreqWarpFIR:45: error: 'audioTestMeasurement' was not declared in this scope AudioControlTestAmpSweep_F32    ampSweepTester(audio_settings,audioTestGenerator,audioTestMeasurement);
* File "WDRC_8BandIIR_wBT.ino.hex". 74520 bytes, 116% used
* File "WDRC_8BandIIR_wBT_wAFC.ino.hex". 89688 bytes, 139% used
* File "WDRC_8BandFIR_wBT.ino.hex". 85568 bytes, 133% used
* File "WDRC_8BandFIR.ino.hex". 85056 bytes, 132% used
* File "WDRC_3BandIIR_wBT_wAFC.ino.hex". 88672 bytes, 137% used

I haven't dug into the actual compile error, but the others all claim they resulting sketch is too large to fit onto the chip.
Perhaps this is something to do with the Teensyduino 1.46 update?

chipaudette, 2019-08-05:
I'm sorry for the basic question back to you...but In the Arduino software, under the "Tools" menu, did you change your "Board" to "Teensy 3.6"?  

Chip

chipaudette, 2019-08-05:
Also, I fixed the error with the WDRC_8BandFreqWarpFIR as of commit f9d82558f8304a24e2d246bbf261f2f0a3855493

Good find!

As for the memory issue, please see my comment above.  If you're not set for Teensy 3.6, it is very easy to see how it thinks you won't have enough memory.

Chip

shapr, 2019-08-06:
I fired up the Arduino IDE and checked my settings (I'm using WDRC_8BandIIR_wBT since I believe that's the default firmware).
I do have the "Teensy 3.6" board selected (basic questions are fine, I make mistakes all the time!)
In the process, I discovered the Arduino menu has "Tools" -> "Optimize" and the "faster" option was selected. Once I selected "smallest", the Teensyduino loader stops complaining.

Should the code fit on the board with the "faster" option selected? 
I also tried "fast" and "fastest" but once the hex file reaches the teensyduino loader I get "too large".

chipaudette, 2019-08-08:
I finally found someone here who uses Ubuntu.  We're going to try to recreate your problem...

jdw-creare, 2019-08-08:
@shapr , I was able to reproduce your issue.  I've reported it to the Teensy maintainers here: https://github.com/PaulStoffregen/Arduino-1.8.9-Teensyduino/issues/3

closed 2019-09-28 by wheresaddie