Tympan/Tympan_Library #21: External Delay

finnwoodman, 2020-07-15:
Hej Tympan Crew, 
I just stumbled across your incredible audio board. It looks like the perfect alternative for what I am doing with the teensy audio board right now. I was wondering whether Tympan has the capabilities to use something to Teensy Audio's AudioEffectDelayExternal. This allows creating longer lasting delays by using additional memory – either placed on the pcb directly or by incorporating a custom memory board. 
Best regards, Max

biomurph, 2020-07-15:
@finnwoodman @chipaudette 
So glad you found us!
I'm pinging Chip, who will be able to help answer.
Tympan works with the teensy audio library, and we have an SD card slot on board, if that is fast enough for your needs. If SD is too slow, we have breakout pins for the SPI bus, so you can lash up a flash memory chip or other.

What is your project and speed/memory needs?

finnwoodman, 2020-07-15:
Thank you for your fast reply. I guess SD card will be to slow as I am processing live sound. So, what I want to create is a 2sec echo loop of a microphone input. The overall goal is creating a kind of simple loop machine.

I managed to create such a "Machine" using the teensy audio board and external ram. Nevertheless I am facing a lot of noise. Tympan looks promising regarding noise and co...

biomurph, 2020-07-15:
Great video!
And great machine.

Adafruit has a couple flash breakouts. Here's one
https://www.adafruit.com/product/1895
It's I2C, and we've broken out the I2C on the Tympan. Note the lower left corner of the Tympan D [schematic](https://github.com/Tympan/Tympan_Rev_D_Hardware/blob/master/TympanRevDschem.pdf) which corresponds to the pinout looking down on the Tympan board. There are lots of development modules that use SPI as well if you really need speed.

I've had parts on my bench for a while to build an effects pedal with a Tympan D. Sparkfun has a nice [kit](https://www.sparkfun.com/products/13124) and I plan to use a flash memory chip in it. Just need the time to work out the power supply...

chipaudette, 2020-09-09:
@finnwoodman, currently the hardware has no place to add external memory.  So, you have the same memory as a Teensy 3.6.

If you need additional RAM, you'll need a product like the blackaddr Teensy Guitar Audio Shield: http://blackaddr.com/products/.  It has options that you can include when you place your order to add a 4Mbit or 8Mbit memory IC.