Tympan/Tympan_Library #23: PlatformIO

jcamins, 2020-09-06:
I know GitHub issues aren't really supposed to be for questions, but this is a question specifically about GH so it doesn't feel like it really belongs on the forum.

I just spent some time getting VSCode + PlatformIO working with the Tympan. I found myself wishing repeatedly that it had been documented here, so I thought I would contribute that documentation now that I have figured it out. I'm not sure where it would belong, though: `readme.md`, `docs/` folder, or wiki. Would instructions be useful to others, and, if so, where would you like them?

wheresaddie, 2020-09-07:
this is a great suggestion, thanks for submitting it. I would suggest it live in docs.

biomurph, 2020-09-08:
@jcamins if it is a 'how to' or a tutorial it could live in the wiki. 
Do you already have a draft of your documentation that we could take a look at? Or you could submit a pull request.
Our wiki also discusses PlatformIO, so that seems a likely spot.

jcamins, 2020-09-09:
@biomurph it's a pretty straightforward tutorial, but it somehow took me... more hours than I care to admit... to figure out step 5. I'll do a PR to `docs/` and after we've resolved any comments that might come up you can decide whether you want to merge it or stick it in the wiki. I'm happy with either!