Tympan/Tympan_Library #28: WDRC for left and right ear separately

yuan9778, 2020-10-03:
We tried to implement WDRC for left and right ear separately. That's 2 WDRC in total. Then RevD stop working, no output. I'm wondering why. The algorithms loaded are tested and should be no problem, so is it because the hardware (memory, storage or CPU) of RevD can't run such a heavy-duty algorithms? 

Thanks in advance!
Sanqing

chipaudette, 2021-06-10:
The RevD has enough CPU to do something like 8 channels of FIR filtering + 8 channels of WDRC.  The 8 channels can be used all on ear or you can divide them among the two ears.

The up-coming RevE should be able to do about 3x as many channels of this kind of time-domain processing.  As we build toward the RevE release, I hope to expand the examples to include some stereo examples.

Chip

yuan9778, 2021-07-05:
Thanks much for your reply!

closed 2021-07-05 by yuan9778