Tympan/Tympan_Library #32: Using Tympan Library 32bit audio output code

dchaplinsky, 2021-01-04:
Hello!

I'm experimenting with teensy based synth and I'd like to take an advantage of 32bit/higher samplerates that Tympan Library provides.

Is there any clues how I can use it solely for this purpose? So far I'm even failing to build a simplest example :( Maybe you can give me a hint on how I can extract audio output code only, as for my needs I don't need all the DSP related to hearing aid and peripheral (like SD).

I'd be super grateful if you can help me with that

chipaudette, 2021-01-05:
Hi!

This kind of discussion is much better for the forum, rather than the GitHub issue tracker.  So, I started a thread over at the forum here: https://forum.tympan.org/t/examples-of-output-only-no-input/283

Chip

closed 2021-01-05 by chipaudette

dchaplinsky, 2021-01-05:
Thanks, William!
I'll post details of my project to the forum and explain what I'm looking
for tomorrow!

Have a good day!

dchaplinsky, 2021-01-06:
I wrote couple of replies at the forum, would be super grateful, if you'd have a minute to glimpse through and advice.

Thanks!