Tympan/Tympan_Library #34: USB audio not supported

hriday91, 2021-03-01:
Currently USB audio in/out is not supported by tympan. Any idea on when will it be supported?

chipaudette, 2021-03-01:
Hi!  

USB audio could be supported as long as you always use a 44kHz sample rate with 128 points per audio block.  Its this inflexibility in the underlying Teensy USB audio system that is the problem.

Would 44 kHz / 128 points meet your needs?

Chip

chipaudette, 2021-06-10:
I would like to wrap the Teensy's USB audio class in a Tympan "_F32" wrapper.  I did this before, but it would cause compiler warnings if you didn't set your Arduino IDEA to have the "USB Type" include audio.  Since most people didn't use the USB audio, this was an annoying warning, so I pulled the _F32 wrapped USB audio class out of the Tympan Library

I think that there is a way for our code to listen compiler flags to avoid these warning messages so that we know if USB Audio has been set in the Arduino IDE.  If I can find out how to do that, we can put the _F32 USB audio class back in our library.

It's on the To-do list, for sure.

nithinr07, 2021-06-21:
Hi,

I am trying to add an AudioOutputUSB block to the single-channel WDRC code. However, this code can't really run well on the Rev D with the 44.1kHz and 128 block size setting, so I was wondering if there is any way to send the audio through USB at lower sampling rates.

Thanks!

chipaudette, 2021-06-21:
The USB audio functionality is the exactly the same code as the Teensy Audio Library USB audio code.  That code is limited to 44.1kHz and 128-sampke block size.  Sorry.

Does the FreqDomainDemo example that I linked above work ok?  Is your problem occur only when you use the USB audio with the WDRC?

nithinr07, 2021-06-21:
Yes, as long as the sampling rate is 44.1KHz and the audio block size is 128, it works, but unfortunately, the 8 band WDRC code can't support this on the Rev D I believe.

chipaudette, 2021-06-21:
You are correct.  the 8-band WDRC consumes too much CPU to be performed at 44 kHz.  If you want the USB audio, you will have to use fewer bands.

There is a group in Montreal that is looking to dig deep into the Teensy Tympan USB Audio code (and, therefore, the Tympan USB Audio code) to make it more flexible so that it can support different sample rates.  If they do it, I believe they are planning on doing it in July - September.

nithinr07, 2021-06-21:
Also, just to point out, I noticed that there were distinct "clicks" in the USB output when the sampling rate was 44100 Hz. However, this problem was solved when I changed the sampling rate to 44117 Hz.

chipaudette, 2021-06-21:
Yes, the requirement for 44117 is part of the limitations (and odd behaviors) that come with the Teensy audio library.  

Sorry.

nithinr07, 2021-06-28:
Hello again!

So, I managed to make the Tympan run at 24,000 Hz / 128 sample block size by tweaking the Teensy library (the USB descriptors), but I wasn't able to lower the block size. I noticed that the WDRC code performs best at lower block sizes, so I was wondering if you could provide me with any leads regarding this. 

Nithin

chipaudette, 2021-06-28:
Wow!  Congratulations on gettng the lower sample rate to work over USB!  Very impressive!

As for shorter block sizes, the WDRC is actually more computationally efficient (ie, less CPU consumed) for the larger block sizes like 128.  What works better is that the latency (the delay) of the audio passing through the system is reduced if the block size is made smaller.  

Do you care about latency?

  * If you do NOT care, than your 128-point block size should be great!
  * If you do care, then you should look into changing the Teensy header file [AudioStream.h](https://github.com/PaulStoffregen/cores/blob/adca415331771bbf1560d7a008ccb150b0ec603a/teensy3/AudioStream.h).  Change AUDIO_BLOCK_SIZE from 128 to something smaller (and change the Tympan code to match) and it might work for you.

Let me know how it goes!

Chip