Tympan/Tympan_Library #35: RevE Review of Example Sketches

chipaudette, 2021-06-10:
Prior to the Release of RevE, we need to review the example sketches in the [develop branch](https://github.com/Tympan/Tympan_Library/tree/develop).

Items to do:

  1) Convert all examples from "myTympan(TympanRev::D)" to "myTympan(TympanRev:E)"
  2) Find all sketches that still use the old bluetooth style, which we'll need to convert to BLE
    * Look for code in the loop() function that is "while (Serial1.available())" [which is old style] or look to see if it creates a "BLE" class anywhere [which is the new style]
    * Or, look at the name of the example and see if it has "BT" or "App" in its title
    * We'll need to update all old-style examples to new-style, but that'll require more detailed instructions

Items to test:

  1) Compile for RevE (Teensy 4.1).
    * Confirm that you changed to "myTympan(TympanRev::E)"
    * Confirm that there are no compiler errors.  Note any compiler warnings
    * Confirm that the audio sounds OK
    * If the code uses the volume knob, confirm that it's OK
    * If that code uses the App (and has been updated for BLE), confirm that the App functions with the Tympan
  2) Confirm that it also still compiles for RevD (Teensy 3.6)
    * Temporarily change the code to say "myTympan(TympanRev::D)"
    * Confirm that there are no compiler errors.  Note any compiler warnings
    * Confirm that the audio sounds OK
    * No need to test the volume knob nor the App

hgeithner-creare, 2021-07-16:
The following has been tested using Teensydunio 1.54 and Arduino 1.8.15.

|                           Example                                |   Changed to RevE   |   Check with RevE   |   Check with RevD   |
| ------------------------------------------------- | --------------------- |---------------------- |---------------------- |
|  01-AudioPassThru                                          |            Yes                |              Pass             |                                  |
|   01-BasicGain                                                  |            Yes                |            Pass               |                                  |
|   01-ChangeSampleRate                                  |             Yes               |               Pass            |                                  |
| 01-PrintCPUandMemoryUsage                        |            Yes                |               Pass            |                                  |
|  01-TrebleBoost                                                |                 Yes           |               Pass            |                                  |
|  01-TrebleBoost_wComp                                  |                 Yes           |               Pass            |                                  |
|   02-BluetoothAudio_PassThru                         |                                 |                            |                                   |
|   02-ControlViaSerial                                        |                                 |                            |                                   |
|   02-ControlViaSerialAndBluetooth                  |                                 |                            |                                   |
|   02-DetectExtMic                                             |                             |                            |                                   |
|   02-EchoBLE                                                     |                             |                            |                                   |
|   02-MyAudioAlgorithm                                   |                             |                            |                                   |
|   02-OutputToneWithSteppedamplitude         |                             |                            |                                   |
|   02-RenameTympanBT                                    |                             |                            |                                   |
|   02-SDWavPlayer                                             |                             |                            |                                   |
|   02-SDWriting_01_StereoAudio                       |                             |                            |                                   |
|   02-SDWriting_02_RemoteControlled             |                             |                            |                                   |
|   02-SoundLevelMeter                                     |                             |                            |                                   |
|   02-SoundLevelMeter_2Chan                          |                             |                            |                                   |
|   02-TympanPDMMics                                      |                             |                            |                                   |
|   03-WDRC_FIR_8Band                                     |                             |                            |                                   |
|   03-WDRC_SingleBand                                    |                             |                            |                                   |
|   04-FormantShifter_FD                                    |          Yes                  |          Pass                 |                                   |
|   04-FrequencyCompressor_FD                        |            Yes                 |           Pass                |                                   |
|   04-FrequencyShifter_FD                                 |               Yes              |         Pass                  |                                   |
|   04-GenericFreqDomainExample_FD               |         Yes                    |          Pass                 |                                   |
|   04-LowpassFilter_FD                                       |          Yes                   |          Pass                 |                                   |
|   04-NoiseReduction_FD                                   |          Yes                   |            Pass               |                                   |
|   05-WDRC_3BandIIR_wAFC                              |                             |                            |                                   |

chipaudette, 2021-07-20:
Regarding the errors/warnings upon first compilation...

I fixed the "jump to case" warning and its associated note about "crosses initialization".  The problem was that I created a variable ("int pinVal") within one of the cases in a switch-case block of code.  While that's sorta fine, it's best practice to employ curly braces within the case to explicitly limit the scope of pinVal.  Once I added the curly braces, this warning went away.

I also got rid of the unused new_val variable, making that warning go away.

I don't know what to do about the type-punned pointer (that was originally created by @kxm-creare).  I don't know if we should avoid employing this technique, or if there is some simple fix to suppress the compiler warning.

Not shown in your report is that there is that the compilation can fail due to some minw (or is it mingw?) issue.  It's like there are too many items getting compiled it gets confused.  If you compile again (when it uses some cached items from the first compile) it compiles fine.  I wish I knew how to avoid that problem, but I don't.  :(

chipaudette, 2021-07-20:
Regarding the message printed with the Teensy 4.1...

That's a fine message.  That's the way that Teensyduino 1.54 as expanded its reporting of FLASH and RAM usage.  The Teensy 4.x has a more complicated memory structure than the Teensy 3.x.  The Teensy 4.x has two RAM areas.  They expanded the reporting to be more explicit.  I don't know what the details of the message mean, but I guess it's a good thing.

Either way, your screenshot reflects normal operation (I believe).  I think that this means that it's good.

chipaudette, 2021-07-21:
@hgeithner-creare , for the example "02-BluetoothAudio_PassThru", it actually has not bluetooth commands being sent back and forth between the Tympan and the TympanRemote app.  Instead, it enables you to hear audio being streamed by your phone from your phone's music player or podcast player or youtube (or, the actual phone part of your phone...crazy).

In your phone's bluetooth control panel, you should be able to connect to the Tyampn manually.  Once you do that, you can go to any app (or a browser and then to YouTube) and play something with audio.  If you plug headphones in the **Tympan**, you should hear your phone's audio through the Tympan.  The volume buttons on your Phone should affect the phone's audio coming out of the Tympan.

At the same time, you should be able to tap on the case of the Tympan and hear the Tympan's own microphones mixed with the BT audio.

Can you test this sketch on both RevD and RevE and confirm that it behaves like this?  On one of the two devices, it'd also be good to test it on both Android and iOS.  

(Again, you do NOT need the TympanRemote app to do any of this.  This sketch has nothing to do with the TympanRemote app.  This sketch relies on the built-in capabilities of the BT module on the Tympan to talk with phones.  So, hopefully, it just works.  Hopefully.)

chipaudette, 2021-07-21:
02-SDWriting_02_RemoteControlled has been updated for BLE.  

@hgeithner-creare Please test.  Lights blink slow (1/sec) when not recording.  Lights blink fast (10x / sec) when recording.  That's a good enough test.

chipaudette, 2021-07-21:
02-SoundLevelMeter_2Chan has no bluetooth interaction to the App.  So, please remove the comment to "Configure for BLE".

Instead, we should consider dropping it.  Does it really add anything when we already have the single-channel SLM example?

chipaudette, 2021-07-21:
@hgeithner-creare, now that we have updated the BLE class for RevD, can you retry the single-channel SLM example ("02-SoundLevelMeter").  It works on my RevE.

Here's a test method: 
* After loading the sketch onto the Tympan, connect via the App.  Make sure it connects.  
* Tap on the icon to get over to the first page of buttons.  Tap on C-Weigiht or FAST and confirm that the button toggles over to green.  Switch back to the originals.
* Swipe to get to the second page of buttons, which is actually a graph.  Tap on 'Start plot'.  Do you see data points!  Snap your fingers or clap near the mics.  Does the graph leap up!

hgeithner-creare, 2021-07-22:
> @hgeithner-creare , I see that you flagged the example "02-BluetoothAudio_PassThru" for additional work regarding BLE (because it lacked the usual BLE commands). This, though, is a werdo example in that it employs bluetooth, but not as a means for conveying commands back-and-forth. Instead, it uses bluetooth in a completely different mode. It uses bluetooth just to receive an audio stream from your phone....like you're connected to a BT speaker or something....or like you're connected to a hands-free headset.
> 
> Can you try testing it as a Bluetooth audio device?
> 
> * Open your phone's Bluetooth control panel and connect to the Tyampn manually.
> * Open any audio-related app on your phone (or a browser and then to YouTube)
> * Play something with audio.  I
> * Plug headphones in the **Tympan** and you should hear your **phone's** audio through the Tympan.
> * The volume buttons on your Phone should affect the phone's audio coming out of the Tympan.
> 
> At the same time, you should be able to tap on the case of the Tympan and hear the Tympan's own microphones mixed with the BT audio.
> 
> Can you test this sketch on both RevD and RevE and confirm that it behaves like this? On RevE, can you also confirm that it can receive audio from both Android and iOS?
> 
> (Again, you do NOT need the TympanRemote app to do any of this. This sketch has nothing to do with the TympanRemote app. This sketch relies on the built-in capabilities of the BT module on the Tympan to talk with phones. So, hopefully, it just works. Hopefully.)

hgeithner-creare, 2021-07-22:
@chipaudette I've completed the testing and updated the table above.