Tympan/Tympan_Library #36: RevE Review of Getting Started docs

chipaudette, 2021-06-10:
Prior to release of RevE, we need to update our Getting Started docs.  

There are several getting started docs.  Please review update all:

  * The readme of the [main repo](https://github.com/Tympan/Tympan_Library/tree/master), which you see displayed automatically when you scroll down a little bit, has some quick getting-started info
  * The main repo also has a [wiki](https://github.com/Tympan/Tympan_Library/wiki), though it doesn't say much except point to our other docs.  Please make sure the links work.
  * The full Getting Started docs are [here](https://github.com/Tympan/Docs/wiki/Getting-Starting-with-Tympan)
 

Here are the specific goals of this update:

  * The docs should be approachable and helpful
  * The docs should be applicable to both RevD and RevE
  * The docs do need to be changed to point to the beta release of Teensyduino
    * The beta is downloaded from [the forum](https://forum.pjrc.com/threads/67252-Teensyduino-1-54-Beta-9)
    * Otherwise, the Teensy people's [installation instructions](https://www.pjrc.com/teensy/td_download.html) can be followed, like usual
  * This update of the docs will not involve changing the photos, unless the photos are misleading
  * This update of the docs could involve updating the screenshots...please review the screenshots and recommend whether you think it is necessary

hgeithner-creare, 2021-07-19:
I know we can no longer use [travis-ci.org](travis-ci.org) and have to use [travis-ci.com](travis-ci.com). I'm not sure how you want to go about updating the [Travis CI Integration](https://github.com/Tympan/Tympan_Library/wiki#travis-ci-integration) section.

[image]

chipaudette, 2021-07-19:
It's OK to leave the Travis stuff to someone else, like maybe @kxm-creare, if he's ever able to get back to this.

hgeithner-creare, 2021-07-20:
I can update the Getting Started docs for the TympanRemote App once @kxm-creare releases a new version.

hgeithner-creare, 2021-10-04:
I'm going to close this issue. Let me know if I need to make a separate issue on updating the Travis CI stuff.

closed 2021-10-04 by hgeithner-creare