Tympan/Tympan_Library #37: SD Playback Is Buzzy/Static

chipaudette, 2021-06-10:
The example sketch [SDWavPlayer ](https://github.com/Tympan/Tympan_Library/tree/master/examples/02-Utility/SDWavPlayer)generates lots of buzz and static.  While the system functions, the audio is so bad that it is unusable.

This was first reported on the forum: https://forum.tympan.org/t/sd-wav-file-playback/300/4.  

In the forum post, I show a sketch that relies primarily on the Teensy audio library for SD operations (not the Tympan audio library).  I have confirmed that on my RevD this code produces clean audio.  I have confirmed that switching over to the Tympan library (in my case, the develop branch using the beta #5 teensyduino) results in the audio exhibiting all the buzz/static.

Since it occurs when using the Tympan_Library and not when using just the Teensy Audio library, the problem must be in our code.

chipaudette, 2021-06-10:
Approach:

1) I should digitally pipe the SD audio out the USB audio path to the PC to see exactly what the samples are so that I can see what kind of distortion is being caused.  Maybe half the block is missing?  Maybe just one sample?  Maybe the samples in the block are backwards?  Looking at the raw audio might give some clue about what to look for in the code.

2) If the results of step 1 do not point to an obvious problem, the next step is to re-port the SD player class from the Teensy library into the Tympan library.  My original port was way before T4 existed.  I believe (?) that Teensy has made many changes, perhaps in response to T4 or in response to the (beta) update to the underlying SD classes.

    * Here's our CPP file: [AudioPlayerSD_F32.cpp](https://github.com/Tympan/Tympan_Library/blob/develop/src/AudioSDPlayer_F32.cpp)
    * Here's the Teensy CPP file: [play_sd_wav.cpp](https://github.com/PaulStoffregen/Audio/blob/master/play_sd_wav.cpp)

chipaudette, 2021-06-10:
If anyone wants to volunteer to chase this down, I'm happy to make someone else the assignee!

biomurph, 2021-06-10:
Just for clarity, are you saying that:

1. Teensy Audio Library produces clean audio when piping from the SD card to the USB
2. Tympan Audio Library produces noisy audio when piping from the SD card to the USB

?

I would love to help but I don't know anything about how to read WAV files. Is there a sample number associated with the data? Does it need to be graphed?

chipaudette, 2021-06-11:
Sorry for confusion.  USB has nothing to do with the problem; the problem is SD out to the headphone jack.  I'm presuming that I will also see the bad audio via USB Audio output, but the bad audio is not *caused* by USB audio output.

In truth, I don't think I that I will lean much from my proposed step 1.  I think that I'm going to have to simply blindly re-port the SD code from Teensy form into Tympan form.

chipaudette, 2024-03-19:
I think that i've now fixed this problem.  I can also now record and play at the same time.  I need to create an example for the library.

chipaudette, 2024-03-29:
Example created!  

https://github.com/Tympan/Tympan_Library/tree/main/examples/02-Utility/SD_Card/SdPlay_and_SDWrite

I think that this problem is now solved.

closed 2024-03-29 by chipaudette