Tympan/Tympan_Library #40: Wrong pin on micro specified for AIC reset

eyuan-creare, 2021-07-02:
@wea I believe the wrong Tympan pin is assigned to reset the AIC , not that it matters.  When the AIC boots up it will reset to default anyway.  

Here is the Rev-D board showing pin A8 is connected to nReset
[image]

Teensy 3.6 diagram shows A8 is pin#22
[image]

Reset pin defined as pin#21 on Teensy 3.6
https://github.com/Tympan/Tympan_Library/blob/c44b8c3d6fce48084498e54583e6e4b5d78af186/src/control_aic3206.h#L45

[image]

Digital write called to reset the AIC
[image]

Looks like this was never got updated from  the Teensy 3.2<br>
[image]

biomurph, 2021-07-05:
@eyuan-creare @chipaudette 
Eric you are looking at a reference to the ball array position. 
The pin number is A8 as you say, but that is only reference to the pin number of the BGA
Note in the schematic that I have listed the Teensy pin numbers in BLUE alongside the BGA pin numbers.
[image]

eyuan-creare, 2021-07-05:
@biomurph Thanks for taking a look and straightening me out.  That makes sense.  So is it okay that nReset is assigned to Pin21?

 #define AIC3206_DEFAULT_RESET_PIN 21

biomurph, 2021-07-05:
The nRESET pin is connected to Teensy pin 35 on Rev D

chipaudette, 2021-07-05:
In practice, the nReset gets set by the creation of the Tympan class.  I believe that this overwrites the nReset pin away from the default set by this #define and to whatever pin is specified in Tympan.h for whichever rev you are using.  So, in effect, this #define ends up getting ignored.

Since I created the whole Tympan.h way of handling the diff Tympan revs back for the release of RevD, I bet that #define value is what was used for RevA/RevC.  It's like you found a fossil

closed 2021-07-05 by chipaudette

chipaudette, 2021-07-05:
With your work on the AIC3216, we have to decide whether to fold it into Tympan.h as another variant of Tympan, or whether we create a whole new class.

Given that Tympan.h extends AIC3206, it seems like switching to 3212 breaks that underlying assumption.  We could abstract the interaction with the AIC, but that only makes sense if they offer similar functionality.  I'm not sure that they do.  I think that creating a new AIC3212 class (modeled on the 3206 class...or not) is probably the right path.