Tympan/Tympan_Library #41: Wrong register values for Mic Bias

eyuan-creare, 2021-07-04:
Value for 1.7V mic bias is not offset to the correct bits

Here is the register for Mic Bias (AIC3206), showing bias voltage as bits D5-D4. For example, to set Mic Bias to 1.7V, we need `TYMPAN_MIC_BIAS_POWER_ON   |  0b0001 0000`

[image]

- However in [Tympan Library](https://github.com/Tympan/Tympan_Library/blob/c44b8c3d6fce48084498e54583e6e4b5d78af186/src/control_aic3206.cpp#L162) the values are not shifted to the correct bits, either in the #define or the wrapper function 
```
#define TYMPAN_MIC_BIAS_REG 0x0133 // page 1 reg 51
#define TYMPAN_MIC_BIAS_POWER_ON  0x40
#define TYMPAN_MIC_BIAS_POWER_OFF 0x00
#define TYMPAN_MIC_BIAS_OUTPUT_VOLTAGE_1_25     0x00
#define TYMPAN_MIC_BIAS_OUTPUT_VOLTAGE_1_7      0x01
#define TYMPAN_MIC_BIAS_OUTPUT_VOLTAGE_2_5      0x10
#define TYMPAN_MIC_BIAS_OUTPUT_VOLTAGE_VSUPPLY  0x11
```

- Here is the [setMicBias](https://github.com/Tympan/Tympan_Library/blob/c44b8c3d6fce48084498e54583e6e4b5d78af186/src/control_aic3206.cpp#L287) function.  When 1.7V is selected this results in a value of `0b0100 0001` instead of 0b0101 0000
` aic_writeAddress(TYMPAN_MIC_BIAS_REG, TYMPAN_MIC_BIAS_POWER_ON | TYMPAN_MIC_BIAS_OUTPUT_VOLTAGE_1_7); // power up mic bias
`
So I believe the way it is now, if you select 1.25V that's fine.  But if you select 1.7, 2.5V, or the power supply, then you actually get 1.7V, with bit-0 set (which is reserved) 

@chipaudette perhaps you can double-check this, then I will branch off to make these corrections.  Does this mesh with the behavior you've seen?

eyuan-creare, 2021-07-04:
I don't think it is worth changing for the AIC3212, but I would typically use enums [here](https://github.com/Tympan/Tympan_Library/blob/c44b8c3d6fce48084498e54583e6e4b5d78af186/src/control_aic3206.h#L32).  This allows coders to quickly look up the enum from the function's input argument, and ensures that only the correct range of values are used.  With #defines, it is difficult to search for which define to use, unless you go off an example.
```
//names to use with setMicBias() to set the amount of bias voltage to use
#define TYMPAN_MIC_BIAS_OFF             0
#define TYMPAN_MIC_BIAS_1_25            1
#define TYMPAN_MIC_BIAS_1_7             2
#define TYMPAN_MIC_BIAS_2_5             3
#define TYMPAN_MIC_BIAS_VSUPPLY         4
#define TYMPAN_DEFAULT_MIC_BIAS TYMPAN_MIC_BIAS_2_5
```

to 

```
typedef enum {
    off, 
    Mic_Bias_1_25V, 
    Mic_Bias_1_70V,
   ...
} Mic_Bias_Values_t;
```

then in the function...

```
bool AudioControlAIC3212::setMicBias(Mic_Bias_Values_t biasSetting) {
  switch (biasSetting){
        case Mic_Bias_1_25V:
           aic_writeAddress(TYMPAN_MIC_BIAS_REG, TYMPAN_MIC_BIAS_POWER_ON | TYMPAN_MIC_BIAS_OUTPUT_VOLTAGE_1_25); // power up mic bias
          break;
    }
}
```
<br><br><br>
To take it a step further, I would encode the register value for mic bias voltage like this:
```
typedef enum {
    uint8_t Mic_Bias_1_25V = 0x00, 
    uint8_t Mic_Bias_1_70V = (0x01<<4) ,
    ...
} Mic_Bias_Voltage_t;
```

then in the function:
```
bool AudioControlAIC3212::setMicBias(Mic_Bias_Voltage_t biasSetting) {
    uint8_t regValue = 0;

    // Read in what's there to preserve the other bits 
   regValue  =  aic_ReadAddress(TYMPAN_MIC_BIAS_REG);

   // Mask out the desired bits to change
   regValue &= (~MIC_BIAS_VOLTAGE_BIT_MASK); 
  
    // Write the new register value
    aic_writeAddress(TYMPAN_MIC_BIAS_REG, regValue | biasSetting);
}
```
I think this is much cleaner.

eyuan-creare, 2021-07-04:
In regards to the AIC3212, it uses different mic bias voltage.  Is that an issue with any of our mics?

[image]

chipaudette, 2021-07-04:
We've only ever used the full bias voltage (2.2v?), so these other voltage values have never ended up being an issue.  It's good to get them correct, but it's not yet been revealed as a problem die to always using the full voltage.  

I agree that an enum send better.  Someone else did the initial code where #define was used instead of enum.  Enum seems better.

Chip