Tympan/Tympan_Library #46: BluetoothAudio_PassThru - Cannot Connect RevE

hgeithner-creare, 2021-07-26:
The Tympan RevE does not show up as an audio device on the phone (Android/iOS) when the BluetoothAudio_PassThru sketch is loaded on the Tympan.

Testing instructions:

- Open your phone's Bluetooth control panel and connect to the Tympan manually
- Open any audio-related app on your phone (or a browser and then to YouTube)
- Play something with audio
- Plug headphones in the Tympan and you should hear your phone's audio through the Tympan
- The volume buttons on your Phone should affect the phone's audio coming out of the Tympan

At the same time, you should be able to tap on the case of the Tympan and hear the Tympan's own microphones mixed with the BT audio.

chipaudette, 2021-07-26:
With the v5 firmware for the BC127, I got bluetooth audio to work simply by invoking the sparkfun library, which is discussed in their helpful tutorial here:

https://learn.sparkfun.com/tutorials/understanding-the-bc127-bluetooth-module/all#usage-examples

It says that they used the Advanced Audio Distribution Profile (A2DP) and Audio/Visual Remote Control Profile (AVRCP) profiles to do the remote audio connection.  It's likely that only A2DP is needed to make the audio flow if one is always driving the play/stop/volume from the phone itself (rather than having the Tympan itself command "play" or "stop" of the audio, which is done via AVRCP).

In the v7 firmware, maybe some other protocol is better, but I'd start by looking at how to enable A2DP.

chipaudette, 2021-07-26:
Melody (ie, BC127 Firmware) V7 Documentation for A2DP:
 
[image]

chipaudette, 2021-07-26:
It looks like that "SET PROFILES=0 0 1 0 1 0 0 0 0 0 0 0" is the key line that we need to convey during initialization of the BC127.  What does each zero or one correspond to?  According to the table below, it permits one A2DP sink connection and one AVRCP connection.

[image]

chipaudette, 2021-07-26:
Based on the table above, it looks like "A2DP sink" should already be enabled (has a "2").  Why doesn't it?

chipaudette, 2021-07-26:
Apparently A2DP is an BT Classic feature whereas we have been (recently) only focusing on BLE.  We enable BLE connections by commanding "ADVERTISING ON" but this does nothing to enable the classic BT connections.  Therefore, we must also enable the BT classic modes to be connectable and discoverable via the command:

`BT_STATE ON ON

This is necessary for the V7 firmware.  I'm not sure about the V5 firmware.

hgeithner-creare, 2021-07-26:
I'm not sure that it would be necessary for the V5 firmware because I was able to connect the RevD.

chipaudette, 2021-07-26:
Ray already included a command for "discoverable()" in the BC127 library.  This command only works for V5.  I extended it for V7.  I also altered the BLE class to initialize the module to make it discoverable (BT classic) by default upon startup (and anytime, actually). 

Changes are on feature_BTaudio branch.

This enables a simultaneous BT audio and BLE connection.  The problem is that it still won't receive audio (on my BC127 HD running V7.3HD).  Whenever I start to stream audio to it, the module appears to reset.  :(

chipaudette, 2021-07-26:
It appears that the unit defaults to allow for the GPIO pins to control the system operation, such as when sinking audio.  Since I am not consciously controlling the BC127 pins, maybe it is getting confusing commands?  Let's try disabling the GPIO control via (v7 firmware):

`SET GPIO_CONFIG=OFF 0`

Then write then reset.

Now, the BC127 seems to lock up instead of just resetting.  I have to cycle the power to get the BC127 to respond again.

chipaudette, 2021-07-26:
When you set GPIO_CONFIG to "off", it doesn't disable GPIO control, it merely alters it...one pin is still an input:

[image]

chipaudette, 2021-07-26:
On RevE, PIO 4 is not connected to anything...meaning it is floating...meaning it is high impedance...meaning the BC127 should be fine with driving it as an output.

PIO 0 and PIO 5 are connected to pins on the Teensy.  When I manually configure them so that the Teensy treats PIO 0 as an input (ie, makes it high impedance) and treats PIO 5 as an output set low (ie, forces it to exit data mode), the system no longer freezes when I start streaming audio.   Instead, it merely returns to the same behavior as before...it fails to stream (ie, no audio comes out the headphone jack) and it stops my phone music app after ~3 seconds.

Still in search of a solution...

chipaudette, 2021-07-26:
I'm stumped.  Here's my test method:

I'm using the example sketc below, based on BC127Terminal, but with added '\r' -> '\n' reformatting to allow use of the Arduino Serial Monitor:

```
/*
*	BC127Terminal
*
*	Created: Ira Ray Jenkins, Creare, December 2020
*	Purpose: Demonstrates a serial terminal with the BC127 module.
*
*/

#include <Arduino.h>

usb_serial_class *USB_Serial;
HardwareSerial *BT_Serial;

void echoIncomingUSBSerial(void)
{
    while (USB_Serial->available())
    {
        BT_Serial->write(USB_Serial->read());
    }
}

void echoIncomingBTSerial(void)
{
  char c;
    while (BT_Serial->available())
    {
      c = BT_Serial->read();
      if (c == '\r') c = '\n'; //replace carriage return with newline to make pretty in Arduino Serial Monitor
      USB_Serial->write(c);
    }
}

void setup()
{
    //  myTympan.beginBothSerial();
    Serial.begin(9600);
    Serial1.begin(9600);
    USB_Serial = &Serial;
    BT_Serial = &Serial1;

    Serial.println("Setting GPIO pins that are connected to BC127 assuming SET GPIO_CONTROL=OFF 0");
    pinMode(2,INPUT);  digitalWrite(2, LOW); //RevE PIO5
    pinMode(5,OUTPUT);//RevE PIO0

    Serial.println("Type commands to send to the BC127.  Be sure to use carriage return, not newline...");
}

void loop()
{
    echoIncomingUSBSerial();
    echoIncomingBTSerial();
}
```
Test Steps with RevE (with V7.3HD firmware):

0) The first time you do this, have your phone "forget" your Tympan BT unit
1) Compile and upload the sketch.
2) Open the Serial Monitor
3) Type Status.  Do you see `STATE CONNECTED[0] CONNECTABLE[ON] DISCOVERABLE[ON] BLE[IDLE]`.  If not:
    * type: SET BT_STATE_CONFIG=1 1
    * optionally, type: SET GPIO_CONFIG=OFF 0
    * type: WRITE
    * Type: RESET
4) Pair your phone and Tympan (the first time)
5) In your phone's BT control panel, connect to the Tympan
6) In the serial monitor, you should have seen stuff happen.  Type STATUS to see the connections
7) Open a music player app on your phone and stream music

Assuming that it fails for you, like it did for me...It should run for a ~3 seconds before the app switches to pause.  In the serial monitor, you should see the startup text, like:

```
Sierra Wireless Copyright 2018
Melody Audio V7.3 HD
Build: 1544717993
Ready
CHARGER_DISABLED
```

Bummer.

ijenkins-creare, 2021-08-04:
@hgeithner-creare , Haley, could you create a new sketch from Chip's example above. Compile for the RevE:v7. Once running, bring up the serial monitor, change the newline to carriage return (the dropdown menu left of "Clear output"). All of the following commands are entered via the serial monitor input line.

Enter "STATUS". You should see something like: 
```
STATE CONNECTED[0] CONNECTABLE[OFF] DISCOVERABLE[OFF] BLE[IDLE]
```

If not, what do you see? Try a "RESTORE" command.

Next, toggle BLE mode on with the command "ADVERTISING ON". Should see "OK". Check that the "STATUS" changed.

At this point, you should be able to connect to the Tympan from a phone/tablet via BLE. Assuming that connection is fine, you should see something like:
```
OPEN_OK 14 BLE 63F5A35DDBB5
```
Next, you can enter the command "BT_STATE ON ON", and again check your phone/tablet for bluetooth connections. You should see the same device name, but with *maybe* a different icon? Try to connect to it. Hopefully it spits out something like:
```
OPEN_OK 23 HFP 90B1445F6B2C
ROLE_OK 90B1445F6B2C M
OPEN_OK 20 A2DP 90B1445F6B2C
ROLE_OK 90B1445F6B2C M
A2DP_STREAM_START 20
OPEN_OK 21 AVRCP 90B1445F6B2C
ABS_VOL 21 77
A2DP_STREAM_SUSPEND 20
AVRCP_PAUSE 21
A2DP_STREAM_START 20
AVRCP_PLAY 21
```
Play some music or video files. You won't be able to hear audio, but you should see in the serial monitor that "something" happens. Let it play for a minute, making sure the phone doesn't pause the video/audio and the serial monitor doesn't "stall". You can periodically type "STAUS" just to make sure it's still alive, if you like.

Thanks!

hgeithner-creare, 2021-08-05:
@ijenkins-creare I load the example sketch and the STATUS looks good:

```
08:49:19.989 -> Setting GPIO pins that are connected to BC127 assuming SET GPIO_CONTROL=OFF 0
08:49:19.989 -> Type commands to send to the BC127.  Be sure to use carriage return, not newline...
08:49:19.989 -> STATE CONNECTED[0] CONNECTABLE[OFF] DISCOVERABLE[OFF] BLE[IDLE]
08:49:20.082 -> OK
```
BLE ADVERTISING successful: 

```
08:49:40.582 -> OK
08:49:54.607 -> STATE CONNECTED[0] CONNECTABLE[OFF] DISCOVERABLE[OFF] BLE[ADVERTISING]
```

When I pair the device:

```
08:50:51.202 -> OPEN_OK 14 BLE 5A2AF556C6FE
08:50:51.342 -> PAIR_PENDING
08:50:56.606 -> PAIR_OK D4E6B726F4F8
08:50:56.653 -> BLE_PAIR_OK 5A2AF556C6FE D4E6B726F4F8
```

When I typed "BT_STATE ON ON" I only got "OK" as the return message. I asked for "STATUS" again:

```
08:56:10.854 -> STATE CONNECTED[0] CONNECTABLE[ON] DISCOVERABLE[ON] BLE[CONNECTED]
08:56:10.948 -> LINK 14 CONNECTED BLE 5A2AF556C6FE 23
08:56:10.995 -> OK
```

The device shows up twice in the Bluetooth settings. The icons are different:

[image]

I couldn't get the Tympan to connect - only pair (there is a difference right?) I never heard audio streaming or saw anything in the serial monitor.

chipaudette, 2021-08-22:
We think that this is taken care of by Merge #52 .

closed 2021-08-22 by chipaudette