Tympan/Tympan_Library #49: Rename master to main

ijenkins-creare, 2021-08-17:
We are moving the default master branch to `main`. This transition should be relatively seamless, but I'm creating this issue to distribute information or track anything that arises.

Now that master->main, to update your local copies:
```
git branch -m master main
git fetch origin
git branch -u origin/main main
git remote set-head origin -a
```

ijenkins-creare, 2021-08-17:
[image]

I got the above warning before renaming... I'm guessing we *do* have a GitLab pages. We should verify that is working properly.

ijenkins-creare, 2021-08-17:
Confirmed, GitHub Pages is functional.

Anyone else, let me know if there are any other issues. Otherwise, I'll close and we'll consider this a success.

kxm-creare, 2021-08-17:
No issues on my end.  If you use SourceTree:

* Find the "master" branch in the left-most pane of SourceTree, under "Branches"
* Right-click on "master" and "Rename master..." to "main"
* Right-click on the newly-named "main", "Track remote branch" set to "origin/main"
* At the top of SourceTree, click "Fetch", make sure "Prune tracking branches no longer present on remotes" is checked, and click "OK"

hgeithner-creare, 2021-08-17:
The links to the Tympan_Library "master" branch still work in the forum! Github displays a little note at the top of the browser letting you know that the "master" branch has been renamed to "main".

chipaudette, 2021-08-22:
Looks good.  This seems done.  Closing.

closed 2021-08-22 by chipaudette