Tympan/Tympan_Library #58: App: addCard_audioSource not displaying

eyuan-creare, 2021-09-23:
In [TrebleBoost_wComp_wEarpieces_wApp](https://github.com/Tympan/Tympan_Library/tree/main/examples/09-Earpieces), the first page does not show options for selecting audio source.

Here is the code as of b173ad9c67148aca65d93873728ac156479422ee
```
TR_Card* EarpieceMixer_F32_UI::addCard_audioSource(TR_Page *page_h) {
  if (page_h == NULL) return NULL;
  TR_Card *card_h = page_h->addCard("Audio Source");
  if (card_h == NULL) return NULL;
  String prefix = String(quadchar_start_char)+String(ID_char)+String("x");
  card_h->addButton(F("Digital Earpieces"),       prefix+"d", F("configPDMMic"),   12); //label, command, id, width
  card_h->addButton(F("Analog: PCB Mics"),        prefix+"w", F("configPCBMic"),   12); //label, command, id, width
  card_h->addButton(F("Analog: Mic Jack (Mic)"),  prefix+"W", F("configMicJack"),  12); //label, command, id, width
  card_h->addButton(F("Analog: Mic Jack (Line)"), prefix+"e", F("configLineJack"), 12); //label, command, id, width
  card_h->addButton(F("Analog: BT Audio"),        prefix+"E", F("configLineSE"),   12); //label, command, id, width
  return card_h;
}
```

chipaudette, 2021-09-23:
Using release_candidate1, I was not able to recreate this problem.  I successfully get the menu that you were missing.

[image]

hgeithner-creare, 2021-10-04:
I tried this example on main and successfully got the menu.

eyuan-creare, 2021-10-04:
Closing this.  I believe the recent changes fixed this.  Thanks!

closed 2021-10-04 by eyuan-creare