Tympan/Tympan_Library #59: Example: TrebleBoost_wComp_wEarpieces_wApp no output to shield

eyuan-creare, 2021-09-23:
In [TrebleBoost_wComp_wEarpieces_wApp](https://github.com/Tympan/Tympan_Library/tree/main/examples/09-Earpieces), I can hear sound from the earpiece mic to headphones on the Tympan.  However, when I move my headphones to the Earpiece shield headphone jack, I can't hear sound.

I noticed that the quad version of the audio input and output was not used.  I tried changing it to the quad version, but no luck.

existing patchcord
```
AudioInputI2S_F32           i2s_in(audio_settings);     //Digital audio in *from* the Teensy Audio Board ADC.
...
AudioOutputI2S_F32          i2s_out(audio_settings);    //Digital audio out *to* the Teensy Audio Board DAC.
```

I also tried reverting the patchcord to a simple passthru.  Also no luck.
```
AudioConnection_F32         patchcord1(i2s_in,0,earpieceMixer,0);
AudioConnection_F32         patchcord2(i2s_in,1,earpieceMixer,1);
AudioConnection_F32         patchcord3(i2s_in,2,earpieceMixer,2);
AudioConnection_F32         patchcord4(i2s_in,3,earpieceMixer,3);
```

However, the AudioPassthru_wEarpieces does work (earpiece mic sends sound to earpiece shield headphone jack and RIC).

eyuan-creare, 2021-09-23:
Same symptom with the EarpieceManualMixing_wSD.  Sound on the Tympan Headphone jack but not the shield.

Also, the trebleboost example seems to be missing this.  Though perhaps it is now called behind the scenes.
```
   myTympan.enableDigitalMicInputs(true);
   earpieceShield.enableDigitalMicInputs(true);
```

chipaudette, 2021-09-23:
Tested on a RevE using Tympan_Library release_candidate1.  

* Confirmed that AudioPassthru_wEarpieces does work
* Confirmed that TrebleBoost_wComp_wEarpieces_wApp doesn't work (ie, is silent)

I looked at the question of whether `enableDigitalMicInputs` was being called.  When using the EarpieceMixer_F32 class, you use the class's methods to choose between the different inputs.  It does indeed call `enableDigitalMicInputs`.  Here's the trail:

1) In TrebleBoost_wComp_wEarpieces_wApp in setup():
    * It calls `earpieceMixer.setInputAnalogVsPDM(EarpieceMixerState::INPUT_PDM);`
2) In EarpieceMixer_F32, we find setInputAnalogVsPDM:
    * It calls `myTympan->enableDigitalMicInputs(true)` and `earpieceShield->enableDigitalMicInputs(true);`

So, I think that this is not our problem.  Bummer.  That would have been an easy fix.  I'm still on this...

chipaudette, 2021-09-23:
The issue was that it was not using the quad versions of the I2S inptu and output...ie, it needed AudioInputI2SQuad_F32 and AudioOutputI2SQuad_F32.  I think that you maybe flagged this somewhere else.

I also made small memory changes to the audio flow in the EarpieceMixer_F32 class, but I don't think that it had an effect.

Stupid mistake on my part.  How did this not get picked up in testing?  I don't know...

I pushed the changes for EarpieceMixer_F32 and for TrebleBoost_wComp_wEarpieces_wApp to the Tympan_Library, the release_candidate1 branch

chipaudette, 2021-09-23:
@eyuan-creare ,

You said that EarpieceManualMixing_wSD wasn't working for you?

I just tried EarpieceManualMixing_wSD, with an earpiece plugged in, I successfully heard audio from both the Tympan headphone jack and from the Shield's headphone jack.  It worked.

Since this sketch doesn't use the EarpieceMixer, I don't see how my changes could have had any influence.  Perhaps the other code updates that are in the release_candidate1 branch solved the problem?

eyuan-creare, 2021-10-01:
Closing this out as I tried this with the updated library and I can now hear sound from the earpiece mics to the AIC shield headphone output.

closed 2021-10-01 by eyuan-creare