Tympan/Tympan_Library #62: RenameTympanBT.ino bug

biomurph, 2022-01-18:
The RenameTympanBT.ino example sketch appears to have a bug.

The existing code starting on line 38 reads
```
// process to get new Bluetooth name
  USB_Serial->println("*** Processing name info..."); 
  given_BT_name = response.substring(5); //strip off "NAME="
  BT_hex = given_BT_name.substring(13); //get the last 6 digits
  new_BT_name.concat("-"); new_BT_name.concat(BT_hex);
  USB_Serial->print("*** Desired New BT Name = "); USB_Serial->println(new_BT_name);
```

This is not correct because the given_BT_name does not have even close to 13 characters. Running this code causes the Tennsy to reboot, likely because of the "array out of bounds" caused by calling a position in given_BT_name that doesn't exist.
The correct substring start point should be 3. It seems that there is an erroneous 1 in the substring parameter.
Also, the given_BT_name also includes the "OK" that is sent by the module. This has to be stripped off.
Here is the working section of code:

```
// process to get new Bluetooth name
  USB_Serial->println("*** Processing name info..."); 
  given_BT_name = response.substring(5,14); //strip off "NAME=" and "OK"
  BT_hex = given_BT_name.substring(3); //get the last 6 digits
  new_BT_name.concat("-"); new_BT_name.concat(BT_hex);
  USB_Serial->print("*** Desired New BT Name = "); USB_Serial->println(new_BT_name);
```

chipaudette, 2022-01-18:
To help document this issue:
* Which Tympan Rev were you using? (presumably E?)
* Which BT module were you using? (HD or non-HD?)
* Which firmware is running on the BT module?  (presumably V7.3?)

Also, be aware that renaming the BT module has lost a lot of its appeal.  The problem is that the full Bluetooth name is no longer displayed on most phones now that the Tympan is using BLE instead of BT Classic.  When the phone senses a BLE-type connection, the phone (my Android phone, at least) displays the BLE name.  The BLE name is the "short" name.  It's so short (8 characters? or only 6?) that you can't really give it a meaningful name. 

Yes, even with such a short name, you could call it "TympanE", but this gives up on including any portion of the BT hex address.  If you've got more than one Tympan floating around, you won't be able to tell the difference.  Obviously, for me, this would be really bad.  Since I want to keep a bunch of those hex characters, I'd only really have 2 characters of the short name that I could customize...which didn't seem worth it.

Your preferences are likely different...I just wanted to make sure that you were aware of the situation with BLE causing the "short" name to be the relevant one.

biomurph, 2022-01-19:
Using Tympan Rev E
BT Module is HD
Running Melody v7.3

Thanks for the clarification on module name use.
The bug causes the Teensy to restart. 

The included [.h file](https://github.com/Tympan/Tympan_Library/blob/main/examples/02-Utility/RenameTympanBT/renameBT_RevD_RevE.h) for RevD and RevE also has a section that sets the new_BLE_name, which is `TYM1234` where the last four are taken from the hex address, so there is some attempt at visible difference appearing in the advertisement. Here's the section . 

```
// set the short BLE name
  new_BLE_name.concat(BT_hex.substring(2));  //get the last 4 digits
  USB_Serial->print("*** Desired New BLE Name = "); USB_Serial->println(new_BLE_name);
  BT_Serial->print("SET NAME_SHORT="); BT_Serial->print(new_BLE_name); BT_Serial->print('\r'); delay(1500); echoIncomingBTSerial();
  BT_Serial->print("WRITE"); BT_Serial->print('\r'); delay(500); echoIncomingBTSerial();
  BT_Serial->print("GET NAME_SHORT"); BT_Serial->print('\r'); delay(500);  echoIncomingBTSerial();
  USB_Serial->println("*** BLE Name setting complete.  Did it return the desired name?");
```