Tympan/Tympan_Library #67: Using the AIC CODEC Shield as Main sound processor

zvikapika, 2023-04-12:
Hi,
Due to Tympan Rev E being sold out, I have been trying to connect the AIC CODEC Shield directly to a Teensy 4.1 board, with no luck. 
The connections I have made are according to the documentation and the Rev E PCB, as following:

Shield                                 |                       Connected To
**First row (from left)**
1.8V                                                                Voltage regulated 1.8V (stepped down from on 5V of Teensy)
IN1 RIGHT                                                      NC
IN1 Left                                                          NC
LINE OUT LEFT                                               NC
LINE OUT RIGHT                                            NC
GND                                                               Teensy GND

**Second row**
MIC BIAS                                                       NC
MCLK                                                            Teensy D23
BCLK                                                             Teensy D21
WCLK                                                            Teensy D20
GND                                                              Teensy GND

**Third row**
PDM CLK                                                       NC
PDM DAT                                                       NC
SDO                                                               NC
SDI                                                                 NC
SCK                                                                NC
CS                                                                  NC

**Fourth row**
3.3V                                                              Teensy 3.3V
DOUT                                                            Teensy D6
DIN                                                                Teensy D32
SDA2                                                             Teensy D18 (Wire2 changed to Wire, also tried Wire1)
SCL2                                                              Teensy D19 (same)
GND                                                              Teensy GND

Also, visual legend of the connections made can be found here:
https://www.dropbox.com/s/wg9cwudwld1ig52/AIC%20CODEC%20Shield%20Connections.png?dl=0

Upon loading the **AudioPassThru** sketch, I am getting no sound on the headphones connected to the shield, and serial monitor shows the following error messages:

AudioPassThru: Starting setup()...
AudioControlAIC3206: Received Error During goToPage(): Error = 2
AudioControlAIC3206: Received Error During goToPage(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During goToPage(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During goToPage(): Error = 2
AudioControlAIC3206: Received Error During goToPage(): Error = 2
AudioControlAIC3206: Received Error During goToPage(): Error = 2
AudioControlAIC3206: Received Error During goToPage(): Error = 2
AudioControlAIC3206: Received Error During goToPage(): Error = 2
AudioControlAIC3206: Received Error During goToPage(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During goToPage(): Error = 2
AudioControlAIC3206: Received Error During goToPage(): Error = 2
AudioControlAIC3206: Received Error During goToPage(): Error = 2
AudioControlAIC3206: Received Error During goToPage(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During goToPage(): Error = 2
AudioControlAIC3206: Received Error During goToPage(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During goToPage(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During goToPage(): Error = 2
AudioControlAIC3206: Received Error During goToPage(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During goToPage(): Error = 2
AudioControlAIC3206: Received Error During goToPage(): Error = 2
AudioControlAIC3206: Received Error During goToPage(): Error = 2
AudioControlAIC3206: Received Error During goToPage(): Error = 2
AudioControlAIC3206: Received Error During goToPage(): Error = 2
AudioControlAIC3206: INFO: Read Page.  Page: 0 Reg: 27.  Failed to go to read page.  Could not go there.
AudioControlAIC3206: Received Error During goToPage(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During writeRegister(): Error = 2
AudioControlAIC3206: Received Error During goToPage(): Error = 2
AudioControlAIC3206: Received Error During goToPage(): Error = 2
AudioControlAIC3206: Received Error During goToPage(): Error = 2
AudioControlAIC3206: Received Error During goToPage(): Error = 2
AudioControlAIC3206: Received Error During goToPage(): Error = 2
Setup complete.
Running...

If there is any way we can get this to work, I'd appreciate any help at all. This project is done during a hackathon in Nigeria taking place this week, and the team is very excited to see the project work. However, so far we haven't been able to do so...
Thanks in advance for any pointer or help,
Zvika

chipaudette, 2023-04-12:
Hello Zvika!

The type of errors that you're seeing tell me that the AIC3206 is not responding to the Teensy.  Typically, the problem is that the "reset" pin of the AIC3206 is not connected back to the processor (to the Teensy).  So, I looked at the connection map that you provided.  I think that I see the problem.  Or, at least, I see one problem.

```
**Second row**
MIC BIAS                                NC  -> No, connect to D36 (or to Gnd)
MCLK                                    Teensy D23
BCLK                                    Teensy D21
WCLK                                    Teensy D20
n_RESET_1                               You forgot this one!  This is critical!  Connect to D26!
GND                                     Teensy GND
```

I have not fully checked your other connections, but try making the changes above and let me know!

Chip

chipaudette, 2023-04-12:
Below is an explanation of the two changes above, for anyone that cares...

Regarding the MIC_BIAS connection:

- This change is not critical.  It is not the cause of your error messages.  But, once you get the system working, this change will help your system's audio inputs work correctly.
- The "MIC_BIAS" connection is an **input** to the AIC Shield, not an output from the shield.  It is intended to be connected back to a GPIO pin on your microcontroller.  The microcontroller sets it HIGH or LOW.  If it is left floating ("NC"), I don't know what will happen.
- The purpose of this line is to control whether mic bias is applied to the right input channel (IN_3_R).  The AIC itself must be commanded (via software command) to generate the mic bias voltage.  By default, though, the mic bias is only applied to the left channel (IN_2_R).  To have the AIC mic bias voltage also be applied to the right channel (IN_3_R), you must pull the "MIC_BIAS" pin high.  There is a command in the Tympan_Library to help with this.
- If you never want mic bias, and you do not want to connect it back to the microcontroller, you can simply tie this pin to GND

Regarding the n_RESET_1 pin:

- This change is critical.  The AIC will never work without making this change.
- The "n_RESET_1" pin is an input to the AIC Shield.  It must be connected back to the microcontroller.  It is used as part of the initialization and configuration process for the AIC (which is handled by the Tympan_Library)
- The details of how this pin is used is too complicated to describe here.  But, basically, the pin needs to be toggled by through various HIGH and LOW states during the initialization process.  Without a connection to the microcontroller, these HIGH and LOW transitions never happen, which means that the AIC never initializes.  As a result, the microcontroller cannot communicate with the AIC.  The failure to receive any replies from the AIC is what causes the errors that you are seeing.
- In short: you must connect this pin back to the Teensy

Chip

chipaudette, 2023-04-14:
If you're not getting any errors, but you're also not getting any audio, lots could be wrong.  I'd need to see a wiring diagram to help at this point:

* To which AIC pin are you injecting your audio?
*  Where is your audio coming from (a mic or your phone)?  
* Is your audio signal going into the AIC strong enough?  Maybe you should add some gain?
* To which pin are you listening for the audio coming out?
* How are you trying to listen to it (ear phones,  loud speaker,  Line-in on your PC)?

Chip

zvikapika, 2023-04-14:
hi Chip,
thanks again for the quick reply,

attaching the wiring scheme at the end of text, with a picture of the setup

we have tried using both amplified and non amplified stereo lapel microphone pair, connected to the shield via pink PL jack, 
and standard over-the-ear music headphones connected to the shield via black PL jack

if the audio input isn't strong enough, how would I know about it? 
I tried looking for sketches that test the microphone and the speaker separately, but could not find such.

By adding gain, you mean by software or hardware? I tried uploading the basicGain sketch, manually setting gain up to 0.95 (getting message on the serial monitor: servicePotentiometer: Digital Gain dB = 40.0) but still no audible sound was heard on the headphones

As for selecting the input source - you are right, I completely missed that. however I've tried the other two options, same result.

could it be that my earphones are the cause of the problem? Is there a way to test them separately?

Thanks again,
Zvika

Shield                                       Teensy 4.1

**First row (from left)**
1.8V                                         Voltage regulated 1.8V (stepped down from on 5V of Teensy)
IN1 RIGHT                                    NC
IN1 Left                                     NC
LINE OUT LEFT                                NC
LINE OUT RIGHT                               NC
GND                                          GND

**Second row**
MIC BIAS                                     D36
MCLK                                         D23
BCLK                                         D21
WCLK                                         D20
RESET                                        D26
GND                                          GND

**Third row**
PDM CLK                                      NC
PDM DAT                                      NC
SDO                                          NC
SDI                                          NC
SCK                                          NC
CS                                           NC

**Fourth row**
3.3V                                         3.3V
DOUT                                         D6
DIN                                          D32
SDA2                                         D18
SCL2                                         D19
GND                                          GND

[image]

biomurph, 2023-04-17:
@zvikapika 
This is exciting!
I will make a branch of the Tympan Rev E hardware git to modify the board to make this connection.

@chipaudette 
In 'normal' conditions, the AIC Shield is the secondary CODEC, but since this configuration doesn't have the main board CODEC, can the Teensy be connected with the primary pins? That would make this setup with the AIC Shield behave more like the Tympan E with no AIC Shield. Does that make sense?

zvikapika, 2023-04-17:
I am now running the 'OutputTone' sketch, which, if I understood correctly, is supposed to test the headphones / sound generation only, but no sound is heard (I checked the headphones with another audio source and they are OK)

Maybe continuing your question, 
Is the connection of the AIC shield directly to the Teensy according to the above scheme supposed to work just like the Tympan Rev E (without bluetooth)?

Or, could it have to do with the teensy redirecting its audio to the USB due to its definition in the Teensyduino as a USB + MIDI + Audio device?

biomurph, 2023-04-17:
@zvikapika 
This may be the issue that you are having. 
The library may be looking for the main-board CODEC.
Try this:
Use the main I2C bus
-  SCL pin 19
-  SDA pin 18
Use the main reset pin
- nRESET pin 26
Use the main I2S bus
- DIN pin 7
- DOUT pin 8
Use the main MicBias pin
- MIC_BIAS pin 36

The other connections for MCLK, BCLK, WCLK are all correct.
This way, the Tympan library with 'think' that it is talking to the main board AIC.

In the list of connections above, you have pins for SCL2 and SDA2 that are incorrect. Those are for the SCL and SDA.
SCL2 is pin 24 and SDA2 is pin 25.

zvikapika, 2023-04-18:
YES! YES! YES! YES! YES!
I was missing the DIN and DOUT correct wiring, 
now I can get data from the mic and send to the earphones
Thank you so much for your help!
Upon finalizing the PCB, I can post the files in case that would be of any help
Thanks again!
Zvika

biomurph, 2023-04-18:
awesome!

zvikapika, 2023-08-14:
Hi all,
happy so say the PCB works great, here is the gerber PCB design file
https://www.dropbox.com/s/dvrs8dpdlbz85aw/final.rar?dl=1
Feel free to make it available on your site if you wish to

Thanks for all the help! another question, coming up :)
Zvika

biomurph, 2023-08-16:
@zvikapika 
Great news! I am unable to open the file you sent. Do you have your work on github?

zvikapika, 2023-08-16:
Unfortunately not yet, in the process.
But try this, should download as a friendlier format:
https://www.dropbox.com/scl/fo/db8042lsb2tdci6rksucg/h?rlkey=z1zpo9n2cl09diqisckl4z6dy&dl=1
let me know whether it worked
Zvika

biomurph, 2023-08-16:
Yeah, that worked. I will have a chance to look at it in the next days.