Tympan/Tympan_Library #69: Equalizer

zvikapika, 2023-08-14:
Hi all,
We have now entered level 2 of the project, taking the prototype to an audiologist and then to some tests
I was asked to develop an equalizer app that would be used for the self-tune process
(here, in Nigeria, audiologist who know how to do this are quite rare)
The basic idea is to help a hearing impaired client to map his/her own frequency response values, and per (currently) 8 frequency bands find what is the threshold DB per band, so that we can compensate for the bands that need amplification
I searched the examples but could not find a similar example - I figured there must be one, since this is what a hearing aid does, right? So, either I missed it somehow, or it is code that was not included in the examples.
Can you help?
Thanks,
Zvika

biomurph, 2023-08-16:
@zvikapika 
This is totally doable with Tympan, but I don't think that we have any diagnostic code examples.
Full disclosure, I am primarily the hardware engineer on the Tympan project. 
The primary focus for Tympan has been as a development and testing platform for audio compression and other audio manipulation. There may be code for self-fitting hearing aid that could get you most of the way to your result?
Please check the examples that use the bluetooth app. They may be the best suited to build out your project idea.

@chipaudette @eyuan-creare 
Do you have any insight into Zvika's project idea?

zvikapika, 2023-08-16:
Thanks!

I just found this morning a correspondence regarding this need, in which Chip referred to the Equalizer_8BandBiquad example he wrote. I uploaded it but there is a lot of high pitch static noise when I do not speak directly to the microphone, making it impossible to work with. 

I tried attenuating the high pitch frequency channels through the Serial Monitor, as the sketch allows, but noise levels did not  change. 

I also thought it might be an issue with my microphone (using a lapel) but even after changing mic, noise level did not subside. 

@chipaudette 
Thoughts? ideas? 
Thanks!
Zvika

eyuan-creare, 2023-08-16:
Do you receive this high pitch noise when using the straight passthru example?  You could consider recording the sound to SD card before and after the 8-band filter.

zvikapika, 2023-08-17:
No, the pass thru example works fine. but after tweaking half a day with just about everything, I found that the issue was too much gain on the microphone. Simple! when you know it :) Thanks for all the help!
btw the equalizer example has a minor typo:

AudioConnection_F32       patchCord35(gainBlocks[4],0,mixer,4);   //connect gain block back to mixer
AudioConnection_F32       **patchCord36**(gainBlocks[5],0,mixer,5);   //connect gain block back to mixer
AudioConnection_F32       **patchCord36**(gainBlocks[6],0,mixer,6);   //connect gain block back to mixer
AudioConnection_F32       patchCord37(gainBlocks[7],0,mixer,7);   //connect gain block back to mixer

chipaudette, 2023-08-17:
Hi!

Thanks for pointing out the typo in the patchCord names.  I've fixed it in the repo.

Chip

zvikapika, 2023-08-17:
Oh, WOW! 
Dude, I cannot thank you enough for your insightful comments and references

Indeed I am working on a (low cost) hearing aid (for Nigerian market) - hence why we chose to work with a DIY Revision E rather then the earpiece (also, we opted not to use Bluetooth to make things simple for the end consumer). 

I know quite a bit about general microcontroller programming, mechatronics and such, but I am doing my first steps with audio processing. Still, I am currently technical project lead... Next week we will be meeting a local Audiologist with whom we will consult about the process of tuning such a device in more detail, but from an initial Zoom meeting I gathered that:

* The type of hearing impairment we wish to address is characterized by uneven frequency response of the patients, which is asymmetric i.e. not the same in both ears. 

* The process of tuning a device here is composed of playing certain frequencies at ascending volumes and marking at which volume level the patient started hearing the sound. Then, based on each volume registered per frequency, an equalizer is configured (per ear) so that it compensates for the patient's natural hearing impairment

* So, now after successfully modifying your Equalizer example to activate two separate equalizers for Left and Right channels, my goal is, pretty much like you guessed, to create a simple user interface that will allow for the patient to diagnose his/her natural frequency response and use that data as gain parameters for the stereophonic equalizer

Regarding lapels, I guessed that they we not ideal but I figured placing two microphones, one next to each ear, as you said, will give us a starting point. Also, I was looking a bit but count not find directional microphones that are compact - usually they are long and cumbersome, maybe due to reflector shape reasons. Anyway at first stage we thought of 3d printing a small cone reflector, but have not yet arrived at that point - see how that goes. Do you know where we can source a compact directional mic?

Last, regarding compression - it makes perfect sense, I know anything about it, though. I will read your post and try incorporating the example into the equalizer

Thank you so much again, I really appreciate your extensive help. 

Hopefully, this project will make a difference to many people's lives here in Nigeria whose need are currently not catered to at all, due to cost reasons

All the best,
Zvika

chipaudette, 2023-08-17:
Hi Zvika,

Here are some thoughts in response to your comments:

> The process of tuning a device here is composed of playing certain frequencies at ascending volumes and marking at which volume level the patient started hearing the sound. 

Yes, testing to find one's threshold of hearing is very typical.  But, despite being typical and normal, it is still hard to do if you are testing normal hearing people.  Very hard.

The problem is that normal hearing people are able to hear *very* quiet sounds.  Therefore, at the person's threshold of hearing, the test tones will be very quiet.  This has several important implications:

* The test must be conducted in a very, very quiet environment.  Any ambient noise could mask the quiet test tones.  The ambient noise must be quieter than your quiet test tones. 
 This is very hard.  Usually hearing testing is conducted in a sound-proof booth.  If you don't have a such a sound-proof booth, you will not get good measurements of hearing threshold for normal-hearing individuals.  

* The system generating the very quiet test tones must have very low self-noise.  Like with the need to block out ambient noise, the system generating the test tones must not project any audible electronic noise into the test area.  If the system produces any audible hiss, it might mask the very quiet test tones, which will degrade the test.

* If you are wearing a hearing device (such as the Tympan) and you are testing to confirm that you have improved a person's hearing thresholds, the hearing device must also produce very little electronic noise.  Otherwise, the device's own hiss might mask the very quiet test tones.  This will degrade the test.  The hearing device might be very useful in everyday life, where ambient noise levels are higher, but testing down at one's hearing threshold is a very challenging test environment.  You might find that the self-noise of the Tympan (or, more likely, the noise floor of the lapel microphones) might be too high to use a test method that is based on measuring the hearing thresholds.

> Do you know where we can source a compact directional mic?

I think that it is probably best to use omnidirectional microphones.  A directional microphone might be good if you can keep it pointing in the right direction...but which direction is the "right direction"?  In my message, I raised the possibility that you might inadvertently get a directional microphone as your lapel microphone.  I raised this possibility because it is a bad thing.  I would suggest that you stay with omnidirectional microphones.

zvikapika, 2023-08-18:
Thanks, again, for your enlightening comments! 

As the manual tuning of the devices can be anywhere, we cannot rely on a sound proof booth. For the time being we have settled for noise blocking earphones such as the ones used in workshops, that the patient will wear during the calibration process on top of the in-ear headphones/buds. Hopefully that would get us close enough to bring some value.

The generation of the test tones will also be done by the Tympan hardware, so I think it should be relatively noise free, from what I've so far heard.

Thanks again,
Zvika

p.s. I just got the first generation of the 3d printed ear reflectors, you might want to take a look :)
https://www.dropbox.com/s/xhlpnnfk1efk47v/20230818_095406.jpg?dl=0
https://www.dropbox.com/s/drwdgtlofds0az1/20230818_095446.jpg?dl=0
they still need more work ...