Tympan/Tympan_Library #70: unable to compile AudioPassThru example

DD4WH, 2023-11-19:
Hi Chip & Co-workers,

I am highly interested in purchasing a Tympan Rev E. While waiting for stock to become available again, I plan to try some of the code on a Teensy 4.0 with ADC PCM1808 and DAC5102a. 

I have Arduino & Teensyduino installed. The Tympan library was copied into my documents/Arduino/libraries folder.

When I compile, I get the following error:

`In file included from C:\Users\DD4WH\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.57.2\cores\teensy4/WProgram.h:36:0,
                 from C:\Users\DD4WH\AppData\Local\Temp\arduino\sketches\7F807E1D7342C18C321FBAB04B0E9818/pch/Arduino.h:6,
                 from c:\Users\DD4WH\Documents\Arduino\libraries\Tympan_Library\src\AudioFeedbackCancelNLMS_F32.h:17,
                 from c:\Users\DD4WH\Documents\Arduino\libraries\Tympan_Library\src\AudioFeedbackCancelNLMS_F32.cpp:2:
c:\Users\DD4WH\Documents\Arduino\libraries\Tympan_Library\src\AudioFeedbackCancelNLMS_F32.cpp: In member function 'virtual void AudioFeedbackCancelNLMS_F32::receiveLoopBackAudio(float*, int)':
c:\Users\DD4WH\Documents\Arduino\libraries\Tympan_Library\src\AudioFeedbackCancelNLMS_F32.cpp:118:12: error: expected unqualified-id before '(' token
  if (!std::isfinite(x[i])) {
            ^
Multiple libraries were found for "SD.h"
  Used: C:\Users\DD4WH\AppData\Local\Arduino15\packages\teensy\hardware\avr\1.57.2\libraries\SD
  Not used: C:\Users\DD4WH\AppData\Local\Arduino15\libraries\SD
exit status 1

Compilation error: exit status 1`

I tried different Arduino/Teensyduino versions, all give the same error.

Can somebody point me to what I am doing wrong?

Thanks in advance!

Regards, Frank

chipaudette, 2023-11-19:
Hi Frank,

Sorry it's not working for you!  This is one of our (my) newer additions to the library, so I shouldn't be surprised that someone is finding errors.  It's annoying, though, when I'm not getting the same error.

Three thoughts:

* In `Tympan_Library/src/AudioFeedbackCancelNLMS_F32.cpp`, at the top, there is the line where it includes `<cfloat>`.  Could you replace this with `<cmath>` instead?  Does that work?
* What version of the Arduino IDE are you using?  I'm embarrassed to say that I'm still back on 1.8.15.  If you're using 2.x, I have no idea what happens. 
* Can you try compiling for T4.1 instead of T4.0.  Does that make the problem go away?  If so, we'll have a bit more info to solve your problem.

Chip

DD4WH, 2023-11-20:
Thanks Chip for your quick and concise response!

* I replaced `<cfloat>` with `<cmath>` in Tympan_Library/src/AudioFeedbackCancelNLMS_F32.cpp

Now it compiles! BTW: I use Arduino 1.8.15 with TD1.54 :-)

Have to go to work now, will try other things later this week.

Thanks a lot !

Frank

chipaudette, 2023-11-20:
I made the change via commit 33a9c2c

Hopefully this works for everyone else, too!

Chip

DD4WH, 2023-11-20:
Thanks Chip! 
Here, I just deleted include <cfloat>.
Just using include <cmath> instead works well here.

Tried out several examples, very impressive. I really like the PitchShift! I built a Stereo Version which makes spatial hearing of pitch-shifted bird songs possible. Will play around some more in the next days.

closed 2024-05-01 by chipaudette