Tympan/Tympan_Library #77: Sitrep - try every example for compile errors, run errors (Rev F)

stvhawes, 2024-06-16:
New to the project, so while familiarising myself with the code I tried every (non-deprecated) example.  

The compile-fails especially could be my libraries setup rather than any particular issue.  Run errors perhaps because there was a hardware item I should have plugged in...

# Compiles and runs and query-favourite
   ./03-Intermediate/WDRC_FIR_8Band/WDRC_FIR_8Band.ino                       okay  serial monitor in + out
   ./05-FullSystems/WDRC_8BandFIR_full/WDRC_8BandFIR_full.ino      okay  serial monitor in and out
             also tested sweep .. good
   ./06-AppTutorial/BasicGain_wApp/BasicGain_wApp.ino                      okay  serial monitor out

# Compiles and runs, no problemo
   ./01-Basic/BasicGain/BasicGain.ino                            okay  serial monitor out
   ./01-Basic/TrebleBoost/TrebleBoost.ino                        okay  serial monitor out
              CPU Cur/Pk: 0.6%/0.6%, MEM Cur/Pk: 4/6, FreeRAM(B) 499712
   ./01-Basic/TrebleBoost_wComp/TrebleBoost_wComp.ino            okay  serial monitor out
               printGainStatus: Input PGA = 20.0 dB. Compressor Gain (L/R) = 0.0, 0.0 dB.
               CPU Cur/Pk: 2.2%/2.3%, MEM Cur/Pk: 4/8, FreeRAM(B) 499712
   ./01-Basic/AudioPassThru/AudioPassThru.ino                    okay  serial monitor out
   ./01-Basic/PrintCPUandMemoryUsage/PrintCPUandMemoryUsage.ino  okay  serial monitor out
               CPU Cur/Pk: 0.4%/0.4%, MEM Cur/Pk: 4/6, FreeRAM(B) 499712
   ./01-Basic/ChangeSampleRate/ChangeSampleRate.ino              okay  serial monitor out
   ./02-Utility/SD_Card/SDWriter_wApp_wMTP/SDWriter_wApp_wMTP.ino    okay  serial monitor in + out
   ./02-Utility/SD_Card/SDWavPlayer/SDWavPlayer.ino
        * For access to WAV files, please visit https://www.pjrc.com/teensy/td_libs_AudioDataFiles.html.
       repeat loop: Starting audio player: SDTEST1.WAV  
   ./02-Utility/SD_Card/SDWriter/SDWriter.ino
            To Start Recording: Turn the volume pot all the way up.
            To Stop Recording: Turn the volume pot all the way down.
            serviceMicDetect: detected removal of plug-in microphone. On-board PCB mics now active.   
   ./02-Utility/MyAudioAlgorithm/MyAudioAlgorithm.ino                       okay  serial monitor out
   ./02-Utility/RamWavPlayer/RamWavPlayer.ino                              okay  serial monitor out
         Starting sample 'YES'...len = 12000 samples
         Starting sample 'NO'...len = 12000 samples   
   ./02-Utility/ControlViaSerial/ControlViaSerial.ino                      okay  serial monitor in and out
   ./02-Utility/OutputTone/OutputTone.ino                                  okay  serial monitor out
      Changing to start tone, amplitude = 0.01
      Changing to start silence...
      Changing to start tone, amplitude = 0.03
   ./02-Utility/LongAudioBlocks/LongAudioBlocks.ino                        okay  serial monitor out
          : Sample rate = 44100.00 Hz
          : block size = 1024 samples per channel
         CPU Cur/Pk: 0.5%/0.5%, MEM Cur/Pk: 4/6
   ./02-Utility/Bluetooth/Tympan_Test_BLE_nRF52/Tympan_Test_BLE_nRF52.ino  okay  serial monitor in and out
   ./02-Utility/Bluetooth/Tympan_Test_BLE/Tympan_Test_BLE.ino              okay  serial monitor in and out
         serialManager: BLE: Name from module = Tympan-6D4E   
   ./04-FrequencyDomain/NoiseReduction_FD/NoiseReduction_FD.ino             okay  serial monitor in and out
   ./03-Intermediate/Equalizer_8BandBiquad/Equalizer_8BandBiquad.ino         okay  serial monitor in + out
   ./04-FrequencyDomain/FrequencyCompression_FD/FrequencyCompression_FD.ino   okay  serial monitor in and out
   ./04-FrequencyDomain/GenericFreqDomainExample_FD/GenericFreqDomainExample_FD.ino   okay  serial monitor in and out   
   ./06-AppTutorial/TrebleBoost_wComp_wApp/TrebleBoost_wComp_wApp.ino      okay  serial monitor out
   ./06-AppTutorial/TrebleBoost_wApp/TrebleBoost_wApp.ino                  okay  serial monitor out
   ./06-AppTutorial/TrebleBoost_wApp_Alt/TrebleBoost_wApp_Alt.ino          okay  serial monitor out
         still: TrebleBoost_wApp: Starting setup()   needs 'alt'
         *   example, which was "TrebleBoost_wApp".   
         *   As a tutorial on how to interact with the mobile phone App, you should first explore the previous
   ./06-AppTutorial/BasicGain_wApp/BasicGain_wApp.ino                      okay  serial monitor out

   ./08-Earpieces/EarpieceManualMixing_wSD/EarpieceManualMixing_wSD.ino      okay  serial monitor out
   ./08-Earpieces/OutputTone_wEarpieces/OutputTone_wEarpieces.ino                 okay  serial monitor out
   ./08-Earpieces/AudioPassThru_wEarpieces/AudioPassThru_wEarpieces.ino          okay  serial monitor out
add aic shield...  earbuds in shield's output
   ./07-AIC_Shield/TrebleBoost_wComp_Quad/TrebleBoost_wComp_Quad.ino         okay  serial monitor out
   ./07-AIC_Shield/SDWriter_Quad_wApp/SDWriter_Quad_wApp.ino         okay  serial monitor out + in
   ./07-AIC_Shield/AudioPassThru_Quad/AudioPassThru_Quad.ino                 okay  serial monitor out

# Compiles, errors during run
   ./02-Utility/SD_Card/SDWriter_wApp_wLog/SDWriter_wApp_wLog.ino
            Setup: SD configured for 2 channels.
            error: setup: SD begin failed!
            SdError: 0X17,0X0   
[... 103 lines cut ...]
         BTNRH_PresetManager_UI.h:43: warning: unused variable 'old_n_crossover' 
            43 |       int old_n_crossover = old_n_chan-1;
               |           ^~~~~~~~~~~~~~~
         In file included from /Stephen/tympan-repo/libraries/Tympan_Library/examples/05-FullSystems/WDRC_NBand_Stereo_Earpiece_Prescrip_wApp/WDRC_NBand_Stereo_Earpiece_Prescrip_wApp.ino:67:
         /Stephen/tympan-repo/libraries/Tympan_Library/examples/05-FullSystems/WDRC_NBand_Stereo_Earpiece_Prescrip_wApp/BTNRH_PresetManager_UI.h: In member function 'virtual bool BTNRH_StereoPresetManager_UI::processCharacterTriple(char, char, char)':
         BTNRH_PresetManager_UI.h:554: warning: comparison of constant '0' with boolean expression is always false 
           554 |         delay(10); if (flag_fail < 0) { setIncrementChannelsMessage("Failed"); } else { setIncrementChannelsMessage("Complete");}
      Running eg...
         BTNRH_WDRC: Preset_SD_Base: beginSD_wRetry: *** WARNING ***: cannot open SD. sd.begin(SD_CONFIG) = 0
             : Trying again...
         BTNRH_WDRC: Preset_SD_Base: beginSD_wRetry: *** WARNING ***: cannot open SD. sd.begin(SD_CONFIG) = 0
             : FAILED all attempts.  Returning w..

# Deprecated
   ./08-Earpieces/Deprecated/WDRC_XBand_PrescripSave_wApp/WDRC_XBand_PrescripSave_wApp.ino   Deprecated
   ./02-Utility/Bluetooth/Deprecated (RevD, RevE)/ChangeBaudrate/ChangeBaudrate.ino    Deprecated
   ./02-Utility/Bluetooth/Deprecated (RevD, RevE)/BC127LibraryTest/BC127LibraryTest.ino    Deprecated
   ./02-Utility/Bluetooth/Deprecated (RevD, RevE)/BasicGain_wApp_BLEdebug/BasicGain_wApp_BLEdebug.ino    Deprecated
   ./02-Utility/Bluetooth/Deprecated (RevD, RevE)/BluetoothAudio_PassThru/BluetoothAudio_PassThru.ino    Deprecated
   ./02-Utility/Bluetooth/Deprecated (RevD, RevE)/EchoBLE/EchoBLE.ino    Deprecated
   ./02-Utility/Bluetooth/Deprecated (RevD, RevE)/BC127Terminal/BC127Terminal.ino    Deprecated
   ./02-Utility/Bluetooth/Deprecated (RevD, RevE)/BLELibraryTest/BLELibraryTest.ino    Deprecated
   ./05-FullSystems/Deprecated/WDRC_8BandFreqWarpFIR/WDRC_8BandFreqWarpFIR.ino    Deprecated
   ./05-FullSystems/Deprecated/WDRC_3BandIIR_wAFC/WDRC_3BandIIR_wAFC.ino    Deprecated
   ./05-FullSystems/Deprecated/WDRC_8BandIIR_wAFC/WDRC_8BandIIR_wAFC.ino    Deprecated

stvhawes, 2024-06-16:
I don't think there's any urgency since there are awesome working examples.

I'll try my hand at some fixes.  Some are probably already fixed (tests were on 12 June main-branch), and if not then very likely to be addressed before I get to them.

chipaudette, 2024-06-16:
Hi!  Thanks for this work!

Most of your errors are caused by running an example setup for using an SD card in your Tympan, but you must not have an SD card.  If you used an SD card, more than half your errors would go away.

Also, you got an error on the example for renaming the Bluetooth module. As the comments in the code say, and as your own comment say, this example only works for older versions of the Tympan.  It is not indebted for use with Rev F.

You also got a bunch of errors on the Deprecated examples. These are not expected to work. Instead of calling them "deprecated", I probably should have called them "soon to be deleted".

Finally, there are all of the other errors that you found. These are more troubling and surprising to me. Prior to the release of Rev F, I recompiled all of these myself and they all worked at that time.  I'll go back and try to recompile them myself in the next couple days.  

What tool chain were you using to compile and test all of these?  You didn't do them all manually like I did, did you?  You must have done the testing in some automated manner? Did you set up a continuous integration system?  I'm very curious!

Chip

stvhawes, 2024-06-21:
:) manual.  Felt like a good way to sense the scale and breadth of the library, while looking for a few favourite examples to focus on.

Also, I didn't expect you would have been able to watch all the examples on every update.  Automation would be nice but not easy when there's the specific hardware (and past versions of the hardware).  Benefit to audiologists would the key purpose of the library rather than software complexity..

I have yet to fix anything .. but I did make a fork and try out the PR pathway.

stvhawes, 2024-06-23:
Yep, sourcing a microSD did fix the runtime issues, and really brought my Tympan on!

Meanwhile I've run through the compile-issues and worked out a proposed fix for each.  I've tested as best I can but might have misunderstood the purpose and capabilities of an example (so incomplete fix).

Pull request: https://github.com/Tympan/Tympan_Library/pull/78

chipaudette, 2024-06-26:
Nice Pull Request!  Thanks so much for the fixes!

stvhawes, 2024-06-27:
Ty.  Closed.

closed 2024-06-27 by stvhawes