Tympan/Tympan_Library #80: Example SdPlay_and_SDWrite defaults to #IF that demonstrates SD read and write without using actual mic input.

eyuan-creare, 2024-09-03:
https://github.com/Tympan/Tympan_Library/blob/48257f25616b1af9c89cb80f50c2f53681699250/examples/02-Utility/SD_Card/SdPlay_and_SDWrite/SdPlay_and_SDWrite.ino#L84

```
#if 1
  //OPTION 1: Take audio from the SD card, process it, then write it back to the SD card
  String description = String("Read Audio from SD, Process It, and Write Audio Back to SD");

  //Connect the audio objects to each other
  AudioConnection_F32     patchcord1(audioSDPlayer, 0, gain, 0); //connect left channel of the SD WAV file to your processing
  AudioConnection_F32     patchcord2(gain, 0, audioSDWriter, 0); //connect your procssed audio to the left channel of the SD writer
  AudioConnection_F32     patchcord3(gain, 0, audioSDWriter, 1); //connect your procssed audio to the right channel of the SD writer
  AudioConnection_F32     patchcord4(gain, 0, i2s_out, 0);       //connect your procssed audio to the left headphone channel
  AudioConnection_F32     patchcord5(gain, 0, i2s_out, 1);       //connect your procssed audio to the right headphone channel 

#else
  //OPTION 2: Play audio out from the SD card while recording audio to the SD card from the analog audio input
  String description = String("Play Audio Out from SD while Recording Audio from Inputs to SD");

  //Connect the audio objects to each other
  AudioConnection_F32     patchcord1(audioSDPlayer, 0, gain, 0);   //connect left channel of the SD WAV file to your processing
  AudioConnection_F32     patchcord2(gain, 0, i2s_out, 0);         //connect your procssed audio to the left headphone channel
  AudioConnection_F32     patchcord3(gain, 0, i2s_out, 1);         //connect your procssed audio to the right headphone channel
  AudioConnection_F32     patchcord4(i2s_in, 0, audioSDWriter, 0); //connect left input to the left channel of the SD writer
  AudioConnection_F32     patchcord5(i2s_in, 1, audioSDWriter, 1); //connect right input to the right channel of the SD writer 
#endif
```

eyuan-creare, 2024-09-03:
@chipaudette I am having difficulty with my laptop connecting to the Tympan, but I can test this with the #IF 0 and get back to you.

chipaudette, 2024-09-04:
Your laptop won't connect to the Tympan?  Weird.

Sometimes, when a program has been compiled to use "MTP Disk" mode and a new program is being pushed that doesn't use MTP, it can have difficulty accepting the program.  The reverse is also sometimes a problem (if it's got a non-MTP program but you're trying to push a MTP-enabled program to it).  In either case, pressing the Tympan's programming button (the reset button) usually works fine to force the re-programming.

eyuan-creare, 2024-09-04:
Nicole verified that this example can play a wav file from the SD and record a mic to SD card simultaneously.  All that was required was switching the IF statement that was used as a preliminary test.

@chipaudette Shall I go ahead and submit a revision with `#IF 0`?

chipaudette, 2024-09-04:
@eyuan-creare, I think that you have the authority to just go ahead and make the change directly to the repo without a Pull Request.  You can probably even do the change and commit it here on GitHub without having to pull locally / edit / commit / push.

Whatever method you do, please proceed!

closed 2024-09-04 by eyuan-creare