Tympan/Tympan_Library #82: First Time Programming - Teensy Loader stalls at 2%

eyuan-creare, 2024-09-27:
Loading Basic Gain w/App on new Tympan (that @biomurph previously loaded with Basic Gain)
- Stalls at 2%; No blinking light.  Not responding to serial.  No audio output.
[image]

- When I press the Auto button, Teensy Loader downloads firmware successfully.  I verified basic gain is working with app control.
 ```
12:06:42.384 (loader): Auto Button event
12:06:42.385 (loader): Auto mode: enabled
12:06:42.403 (loader): File "C:\Users\eyuan\AppData\Local\Temp\arduino\sketches\5565D6EC6055986BFBB5CC9B3AC67DF1\BasicGain_wApp.ino.hex", 185344 bytes
12:06:42.409 (loader): File "BasicGain_wApp.ino.hex". 185344 bytes, 2% used
12:06:42.418 (loader): begin elf_guess_size
12:06:42.419 (loader): found elf file
12:06:42.421 (loader): read elf files into memory, 4529776 bytes
12:06:42.422 (loader): elf appears to be for Teensy 4.1 (IMXRT1062) (8126464 bytes)
12:06:42.422 (loader): elf binary data matches hex file
12:06:42.425 (loader): elf file is for Teensy 4.1 (IMXRT1062) (id=25)
12:06:42.427 (loader): using hex file
12:06:42.436 (loader): begin operation
```

chipaudette, 2024-09-27:
Did this event happen using the questionable unit?

chipaudette, 2024-09-27:
Were you compiling with MTP enabled (Tools->USB Type->"Serial w MTP (experimental)")?  Or, were you using the basic plain-vanilla USB Serial (Tools->USB Type->"Serial")?

eyuan-creare, 2024-09-27:
The sketchy Tympan TYM-F003, is now working.  

When first plugged in today, no USB ports showed up on the Tympan .  After compiling Basic Gain, it failed to upload.  Then I pressed the program button and it uploaded successfully.  After that, I compiled Basic Gain w/App and it uploaded fine.

chipaudette, 2024-09-27:
We just talked by phone.  You'll keep an eye on them as you work with them over today and Monday. I'll be in on Monday/Tuesday to update their BLE firmware and we can also explore this programming issue.

eyuan-creare, 2024-09-27:
@biomurph If you get the chance, could you try compiling and uploading the example Gain w/App onto a few fresh Tympans?

As I was able to eventually program them (by working thru the Teensy Loader directly) I don't think this is critical.  But good to keep an eye on and also to archive in case users hit this problem.

eyuan-creare, 2024-09-27:
Chip, what version is the ne BT firmware?  I'll log this to Snipe for the (4) units you already updated

eyuan-creare, 2024-09-27:
Something to try if the Tympan does not show up in the Arduino IDE with a serial port.  Taken from Teensy website.  Run the [serial driver installer](https://www.pjrc.com/teensy/serial_install.exe).

```
On Windows, even if the [serial driver installer](https://www.pjrc.com/teensy/serial_install.exe) has run correctly, sometimes Windows will retain old information in its registry and fail to load the driver. The Device Manager will show "USB Serial" with an error, instead of "USB Serial (Communication Class, Abstract Control Model)" with a COM port number assigned.

These actions usually can cause Windows to re-detect the device and begin using the correct driver.

- Run the serial installer program again, or run the Teensyduino installer and use the update driver option on the first step.
- Click that "Update Driver" button in the device manager. If the driver installer has run, Windows should be able to find the driver.
- Plug the cable into a different USB port. The Windows Registry handles each physical port with separate registry entries.
- Do and/all of the above with or without the device connected. Sometimes Windows will do things differently if the device is present or absent.
- ```

biomurph, 2024-09-27:
@eyuan-creare 
When I am preparing and testing Tympans I use Teensy Loader to flash precompiled HEX of BasicGain.ino
After the Loader is done, I see the Tympan appear on the serial port, and I also verify both mics with headphones.

I have found that Teensy units can sometimes require having the program button pressed or occasionally require power down/up.

eyuan-creare, 2024-09-27:
The first time programming 4 fresh units took several cycles of powering on/off, pressing program button.  Perhaps try compiling a sketch and program one unit in the typical fashion, then press the program button.  That sequence did not work on several units.

biomurph, 2024-09-30:
@eyuan-creare 
I popped open a Tympan from my inventory.
It powered up and connected to the serial port just fine.
I compiled and uploaded the BasicGain sketch, which opened Teensy Loader and successfully erased and programmed the Tympan.

One thing to check is to make sure that your USB cable is plugged all the way into the Tympan.
When you plug in, make sure that the charge LED is lit?
I just noticed that I had to push hard to get the USB micro to seat all the way in.

eyuan-creare, 2024-09-30:
@biomurph Thanks for giving this a try. I'll close this issue until I see it pop up again.

closed 2024-09-30 by eyuan-creare

biomurph, 2024-09-30:
@eyuan-creare 
I just powered down the Tympan I successfully programmed, and then powered up and pressed the program button.
Upload by teensy loader was successful. I can’t replicate your problem yet.