Tympan/Tympan_Library #88: Feature: SDWriter: Add option to write INFO metadata to header of WAV file

eyuan-creare, 2025-07-20:
[SDWriter.h](https://github.com/Tympan/Tympan_Library/blob/main/src/SDWriter.h), [SDWriter.cpp](https://github.com/Tympan/Tympan_Library/blob/main/src/SDWriter.cpp)

### Goal
Write channel names and input calibration to the WAV header's INFO<ICMT>Comment chunk.

### Ref
[IBM's specification(https://www.mmsp.ece.mcgill.ca/Documents/AudioFormats/WAVE/Docs/riffmci.pdf) on the RIFF file format that WAV files use.  [INFO chunk](https://www.mmsp.ece.mcgill.ca/Documents/AudioFormats/WAVE/Docs/riffmci.pdf#page=23) described on pg-23 

### Scope 
- Overload [`wavHeaderInt16()`](https://github.com/Tympan/Tympan_Library/blob/383c8b9f7a1dfd8579ebd06e952896f42c019829/src/SDWriter.cpp#L71) with additional argument for an INFO chunk structure.
- After audio is recorded the [`close()`](https://github.com/Tympan/Tympan_Library/blob/383c8b9f7a1dfd8579ebd06e952896f42c019829/src/SDWriter.cpp#L26) method is called to update the file size reported in the header.  This can stay the same since it reads the actual file size.

### Notes
- Per the RIFF format, the LIST <INFO> chunk can go anywhere after the FMT chunk.  But it seems that some WAV readers prefer it before the data chunk, rather than after.
- Per the RIFF standard, LIST <INFO> chunk should contain the following fields.  We are going to write the Comment <ICMT> subchunk.
IARL IART ICMS **ICMT** ICOP ICRD ICRP IDIM IDPI IENG IGNR IKEY ILGT IMED INAM IPLT IPRD ISBJ ISFT ISHP ISRC ISRF ITCH.

eyuan-creare, 2025-07-20:
@chipaudette Do you have a preference regarding the use of union / struct to parse bytestreams?  I am using this to read WAV file headers in a different repo.  I could save this to a separate .h file so both SDWriter and SDReader have access to them.

```
// RIFF header uses union to type pun incoming bytestream.
using Riff_Header_u = union {
  struct {
    uint32_t chunkId;             // "RIFF" as LSB
    uint32_t chunkLenBytes;       // File length (in bytes) - 8bytes
    uint32_t format;              // "WAVE" as LSB
  } header;      
  uint8_t byteStream[12] = {0};   // represented as byte array
};

// FMT header uses union to type pun incoming bytestream.
using Fmt_Header_u = union  {
  struct {
    uint32_t chunkId;             // "fmt " as LSB
    uint32_t chunkLenBytes;       // Expect 16 bytes (this chunk - 4 bytes)
    uint16_t audioFmt;            // 1: PCM, otherwise is compressed
    uint16_t numChan;             // # of audio channels
    uint32_t sampleRate;          // bits per second
    uint32_t byteRate;            // SampleRate * NumChannels * BitsPerSample/8
    uint16_t blockAlign;          // NumChannels * BitsPerSample/8
    uint16_t bitsPerSample;       // 8: 8-bit;  16: 16-bit
  } header;
  uint8_t byteStream[24] = {0};   // represented as byte array
};

// data header uses union to type pun incoming bytestream.
using Data_Header_u = union {
  struct {
    uint32_t chunkLenBytes;                         // Number of audio bytes: NumSamples * NumChannels * BitsPerSample/8
  } header;      
  uint8_t byteStream[CHUNK_ID_LEN_BYTES] = {0};     // represented as byte array
};
```

eyuan-creare, 2025-07-20:
Good point.  I'll explore the other options that c++ offers.

eyuan-creare, 2025-07-20:
@chipaudette In case you reviewed the approach up at the top, I modified it a bit after digging into how the existing SDWriter class works.  Should be a simple mod.

eyuan-creare, 2025-07-20:
@chipaudette This is minor, and probably has no effect as the header is overwritten when closing the file.

When a WAV file is initialized, `openAsWave()` calls `wavHeaderInt16(0)` with a fileSize of 0;  This is due to not knowing how much audio will be written.  However it's a little strange, since it results in the RIFF chunk size being written as a neg number.  

nsamp = (0-44)/(2*1)  = -22 //assume 1-chan
RIFF chunk size = 36 + (-22*1*2) = -8
```
int nsamp = (fileSize - WAVheader_bytes) / (nbytes * nchan);
*(int32_t*)(wheader + 4) = 36 + nsamp * nchan * nbytes;  //what is this?  Why 36???
```
Propose that we change it so that `wavHeaderInt16(44)` is called. This is the file size if writing just the header, no audio data. It results in nsamp = 0;

And since fileSize is specified, instead of RIFF chunk size expressed as this:
```	
*(int32_t*)(wheader + 4) = 36 + nsamp * nchan * nbytes;  //what is this?  Why 36???
```
It is really just the total file size subtracted by the two fields that are not included in the size (8 bytes for "RIFF" and RIFF chunk size).
```
if (filesize>=8) {
    *(int32_t*)(wheader + 4) = fileSize-8;
} else {
    *(int32_t*)(wheader + 4) = 0;
}
```
Note, in the original code, here is where the 36 is from.  (Essentially everything in the header but "RIFF" and RIFF chunk size.)
- Skip "RIFF" (4B) and RIFF chunk size (4B)
- 4 bytes for "WAVE" 
- 8 bytes for "fmt " and fmt size
- 16 bytes for contents of fmt chunk
- 8 bytes for "data" and data size

eyuan-creare, 2025-07-21:
`wavHeaderInt16(0)` returns `static char wheader[48]`.

However, only 44 bytes are written to the file, as `int WAVheader_bytes = 44`; 
```
file.write(wavHeaderInt16(0), WAVheader_bytes)
```
Not sure if I am missing something.  Without the added INFO chunk, audio should start at index 44.  But why is `wheader` 48 bytes long?

eyuan-creare, 2025-08-03:
@chipaudette AudioSDPlayer_F32 currently rejects reading WAV files with more than 2 channels, right?  What was the reason for this?  (Is it due to the RIFF format not having a channel field to specify something like "front rear speaker"?
 https://github.com/Tympan/Tympan_Library/blob/ef3c80c3a6d4ecd9400ed903d5de404211321394/src/AudioSDPlayer_F32.cpp#L890

I believe that AudioSDPlayBack_F32 does not accept the WAV format 0x0003
WAVE_FORMAT_IEEE_FLOAT, just the standard  0x0001 WAVE_FORMAT_PCM?  Are we planning to allow playback of F32 WAV files as well?  I do see that you added the 0x0003 format to SDWriter.
https://github.com/Tympan/Tympan_Library/blob/ef3c80c3a6d4ecd9400ed903d5de404211321394/src/AudioSDPlayer_F32.cpp#L861

I plan to insert the LIST<INFO> metadata chunk before the data chunk, as it seems that some WAV players prefer this.  To accommodate this insertion, I am revising SDWriter so that the buffer for storing the WAV header is not a single, fixed length.  I am about ready to pull in your changes from the branch (https://github.com/Tympan/Tympan_Library/tree/feature/32bit_quad_i2s).  Also, I have code to parse WAV headers that is currently in the OpenHearing project that I may be able to merge with Tympan's AudioSDPlayback_F32.

chipaudette, 2025-08-03:
@eyuan-creare 

In reply to your question, I do believe that AudioSDPlayer_F32 only does 2 channels.  If so, it's because I built it directly from Teensy's AudioSDPlayer, which only does two channels.  So, it wasn't a choice, it was simply expedient and no one has asked for more (until now).  It could certainly be expanded to 4 channels (or whatever).

As for your question of WAV format 0x0003 and the playing back of Float data type, the inability to do this again traces back to the Teensy's AudioSDPlayer, which lacked this capability.  As you noted, we've now added it for writing.  I'd be happy to add it to playback, but I've been asked to hold back in order to reserve resources for higher-priority tasks. 

Finally, when adding the LIST metadata chunk, note I have already done some alteration to the handling of the WAV header when I added support for writing Float32 data type:
* The WAV header is now created by a more-generic method: [makeWavHeader()](https://github.com/Tympan/Tympan_Library/blob/0a61fe3137d40a66e3e078efb34a5b84554e22ab/src/SDWriter.h#L103)
* A buffer for the WAV header is now created at a larger size ([here](https://github.com/Tympan/Tympan_Library/blob/6f47fe5ce487600a6ff8161d7f31f639fccc6939/src/SDWriter.h#L130)) to handle the expected header length for both Int16 and Float32 (you can further enlarge this, along with the line right below it, if needed)
* While the line above allocates space for a larger buffer, it doesn't mean that the full size is actually used.  The header for Int16 data, for example, still needs only 44 characters.  So, the class also keeps track of the size of the header that's actually being used.  The data member keeping track of the length is [here](https://github.com/Tympan/Tympan_Library/blob/6f47fe5ce487600a6ff8161d7f31f639fccc6939/src/SDWriter.h#L126) while the value of this variable is set in makeWavHeader() at [this line](https://github.com/Tympan/Tympan_Library/blob/ef3c80c3a6d4ecd9400ed903d5de404211321394/src/SDWriter.cpp#L159).  
 
To add a LIST metadata chunk, you'll probably have to dig into the new `makeWavHeader()`.  I am absolutely not a master of the WAV header format and options.  As a result, the logic wtihin `makeWavHeader()` is a bit convoluted.  I'm sorry!  While it does work, it could probably be cleaned up.  If you see improvements (that don't break anything), please do the improvements!

If you add the LIST chunk to the header (is it in the header?  or is it added to the end of the file as a footer?), be aware that altering the header will break the Tympan's ability to read its own WAV files.  It is true that my own modifications to add Float32 writing means that the Tympan can't read its own Float32 WAV files, but at least the system can still read its own Int16 WAV files.  

If you're going to alter the header when writing Int16 WAV files, please make sure that you have the budget to also alter the Tympan's reading of Int16 WAV files so that it isn't broken.

eyuan-creare, 2025-08-03:
@chipaudette Thanks for providing background on the limitation of playing F32 WAV files and WAV files with > 2 channels.  Good idea to wait on this until our minimum requirements are satisfied.

My intention was to add the List metadata chunk in the header, as that seemed to be compatible with more players.  (Sorry, I remember looking into this but didn't document my sources). Thanks for the reminder that the Tympan WAV player relies on the data chunk being right after the fmt chunk. In the OpenHearing project, I extended AudioSDPlayer_F32 with a child class that searches for chunkID strings instead of relying on bit position. As the RIFF standard allows the metadata to be placed anywhere after the fmt chunk, this will be more compliant with various WAV editors.  I will try to pull this into the Tympan Library, if I can, using a new feature branch.  Or as a fallback, write to the footer instead.  

@chipaudette To demonstrate writing and reading metadata, should I create a new example sketch based on reading an int16 WAV file and writing an F32 WAV file, similar to SdPlay_and_SdWrite()?

eyuan-creare, 2025-08-03:
### Obsoleting WavHeaderInt16()
I noticed that the 32bit_quad branch comments out the public method, `wavHeaderInt16()`.  Should we retain this for backwards compatibility?  Perhaps with a switch based on whether Int16 writing is enabled?

chipaudette, 2025-08-03:
> [@chipaudette](https://github.com/chipaudette) To demonstrate writing and reading metadata, should I create (2) new example sketches:
> 
> * SDWrite_wMetadata.ino: Write an F32 WAV file w/metadata
> * SDWavPlayer.ino: Read I16 WAV file w/ metadata

Hi Eric,

I do agree that an example with metadata is worthy.  Since metadata probably not the dominant use-case, I would suggest that we only have one example, not two.  Your one example could show both reading and writing.  In fact, that'd be safest as you could write a WAV with metadata as the first part of your example and then you'd be guaranteed that there'd be a metadata-equipped WAV file to demonstrate metadata reading as the second part of your example.

Whatever you choose to do, I agree that "wMetadata" (or whatever) should be part of its name to help a novice user identify which SD example they want to learn (or, in this case, maybe choose to ignore when trying SD for the first time).

eyuan-creare, 2025-08-03:
>handle the case where the class is configured to write Float32 data yet someone goes and invokes wavHeaderInt16().

Within `wavHeaderInt16()`, Could we check `AudioSDWriter_F32::WriteDataType::FLOAT32`?

eyuan-creare, 2025-08-03:
>Your one example could show both reading and writing. In fact, that'd be safest as you could write a WAV with metadata as the first part of your example and then you'd be guaranteed that there'd be a metadata-equipped WAV file to demonstrate metadata reading as the second part of your example.

I agree with a SDRecord_and_SDPlay_wMetadata example, but then I would need to write then read INT16.  If I used separate examples then I could write F32 and read INT16.  Perhaps just modify existing SDWrite and SDPlay examples with a #IFDEF switch to write metadata?

chipaudette, 2025-08-03:
> > handle the case where the class is configured to write Float32 data yet someone goes and invokes wavHeaderInt16().
> 
> Within `wavHeaderInt16()`, Could we check `AudioSDWriter_F32::WriteDataType::FLOAT32`?

Yes, but then what?  The method must return a char array.  What should that array contain if they ask for an Int16 despite being configured for Float32?

eyuan-creare, 2025-08-03:
> Yes, but then what? The method must return a char array. What should that array contain if they ask for an Int16 despite being configured for Float32?

Perhaps, return nullptr and mention in the method description to check for nullptr.  Ensure that our functions check for nullptr.  Unless there is some global system error we can set.

By default, we are using F32 I2S audio now unless a user sets a #define, right?  Can we use a #ifdef to only include `wavHeaderInt16` if reverting to I16 audio?

eyuan-creare, 2025-08-03:
> @eyuan-creare , are you looking to demonstrate writing F32? 
Goal is to demonstrate writing WAV file metadata.  I could update SDWriter.ino to show this, making it clear how to comment out writing metadata.

Perhaps we don't need an example for demonstrating reading metadata (unless we want to show how to access that information).

chipaudette, 2025-08-03:
> Perhaps, return nullptr and mention in the method description to check for nullptr. Ensure that our functions check for nullptr. Unless there is some global system error we can set.
>
> By default, we [...]

Returning `nullptr` also breaks the API, though now it breaks at runtime, which is much more dangerous.  If we return `nullptr` they'll have to change their code but they won't know it until their get weird behaviors associated doing an out-of-bounds membory access.  IMO, that's very frustrating.  At least by moving to `makeWavHeader()`, it fails at compile time and they know that they have to change that one line of their code.

You do know that the new `makeWavHeader()` works for both Int16 and Float32, right?

As for the question of using F32 for I2S, I think that these are two separate questions.  Running 32-bit I2S and writing 32-bit WAV files do not depend upon each other.  I've only included them in the same branch of the repo be we, the initial users, were likely to want to use them together...so i kept them in the same branch.

Is keeping a method called `wavHeaderInt16()` important for your work?  If that specific name isn't important, my conjecture is that everything will work fine if you invoke `makeWavHeader()` instead.  I could absolutely be wrong, but that was my intention.

eyuan-creare, 2025-08-03:
No, I am fine omitting `wavHeaderInt16()`.  Just felt wrong.  But agree that your approach of moving to the new method informs the user at compilation to move to the new method, and limits the risk of real-time overruns.  I'll plan on modifying `makeWavHeader()` then.

chipaudette, 2025-08-03:
> Ideally, opening a file and writing a header would be separate methods to allow more modularity. Then AudioSDWriter_F32 would have a private method with two calls to SDWriter to open the file and write the header.

@eyuan-creare , you're in luck!  `SDWriter` has this capability already!

* If you've been using AudioSDWriter_F32, then you've been using SDWriter's `openAsWav()`.  As you've noted, this both opens the file and writes the WAV header.  It also sets a flag so that, when the file is closed, it goes back and automatically updates the WAV header with the total amount of audio data written.  Convenient!

* But, if you don't want or need this convenience, you can simply open the file using SDWriter's `open()`.  Then, you'd make the Wav header (`makeWavHeader()`) and then tell SDWriter to write those bytes to the SD via `write()`.  No problem!  Just be sure to re-write the header before closing out the file.

Does that achieve the modularity that you want?  I'm happy to put more modularity in there, if that's what you want! (once we have budget, of course)

Chip